prusa3d/PrusaSlicer #8442: Arrange - add parameter to keep given distance from bed edges

Valthol, 2022-06-30:
Arrange function is awesome for sure. When printing many smaller items the arrange funcion often places them mathematicaly like almost at the critical edge of the bed - this often can cause troubles - weaker bed adhesion just at the end, warping due to heat beign lower at the edge etc. 

It would be very usefull to implement a parameter "keep distance from the bed edge" so the user can specify for the print safety whether he wants arrange function to keep given distance from the bed edge. 

This option could just offset the initially given size of the bed, make it smaller and pass it further to Arrange funcion to arrange.

smirgol, 2022-07-08:
This would be really nice to have. 

Another example: 
If I remember correctly, arranging doesn't take into account the required space for the loop (skirt) you usually print around your objects.

Valthol, 2022-11-14:
Have any dev looked at it by any chance? Still it would be a very usefull feature and rather easy to implement.

Valthol, 2022-12-24:
Refreshing this feature request - still it should be very easy to implement and would save hours of clicking for hundrets of people running more printers frequently. Just let us set distance offset from bed edges, like 5mm or so so the parts would be safe inside and not warp due to beign right on the edge of the bed. 

Would love to get any dev comment on this.

kubispe1, 2023-01-22:
Hi, a new arrange function parameter to set min distance from the bed edges is available in 2.6.0.
[image]
Closing as done.

closed 2023-01-22 by kubispe1

ericbjones, 2024-07-10:
"Spacing from bed" is a little confusing.  It would be better as "spacing from bed edge":
https://github.com/prusa3d/PrusaSlicer/pull/13065