prusa3d/PrusaSlicer #8443: The seam algorithm remains confused

ramosglauco, 2022-06-30:
### Description of the bug

Hi. The seam algorithm is creating smooth and curved patterns where there is no need (on sharp corners and straight surfaces for exemple), this was not the case in version 2.4.0. Apparently the algorithm is having trouble making sharp turns too. I'm submitting a lot of comparisons between version 2.5.0-alpha2 and 2.4.0. I hope they finally find a way to improve the seam patterns, as I have suffered from this for a long long time. And please (PLEASEEE!) add an option to make straight lines in the seams. I've been asking this for over 1 year and I've seen many other users ask for the same thing. It's embarrassing to have to use a bug to make a simple line (hold left mouse + esc, then left mouse click again). Thank you.

2.5.0-alpha2
[image]

2.4.0
[image]

---------------------------------------------------------------------------------------

2.5.0-alpha2
[image]

2.4.0
[image]

---------------------------------------------------------------------------------------

2.5.0-alpha2
[image]

2.5.0-alpha2
[image]

2.4.0
[image]

### Project file & How to reproduce

[porta-palhetas-caveira-geometrica-mohawk.zip](https://github.com/prusa3d/PrusaSlicer/files/9020123/porta-palhetas-caveira-geometrica-mohawk.zip)

[suporte-vertical-de-placa-de-video-jacked-v1-parafuso.zip](https://github.com/prusa3d/PrusaSlicer/files/9020129/suporte-vertical-de-placa-de-video-jacked-v1-parafuso.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2

### Operating system

Win11

### Printer model

Artillery Genius

DragonFinlay, 2022-09-15:
I was able to reproduce all of these in 2.5.0 final, the seams look exactly the same.
We are working on improving the seams with Arachne in general and we will probably keep yours and other similar issues (#8443, #8512, #8841) open until we can confirm this is fixed. Thanks for the screenshots and 3MFs!

About painting in straight lines, we are keeping #6557 open until we can implement it.

Jan-Soustruznik, 2024-09-16:
Hello, @ramosglauco , Could you check the seam placement, with new PrusaSlicer release, 2.8.1-rc1 (link to download: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1-rc1). 

Seam algorithm was enhancement now the placement of seams with specific object geometry should be improved. "Rear" placement was changed back before slicer 2.6.0.

bryn51, 2024-09-16:
So this issue is about the seam lines not being straight ? But another report about the seam lines being too obvious closed in favour of this one?

Seems (obvious pun) that whats needed is deployment of Scarf Seams to make the seam lines less obvious and of course if they remain visible then they ought to be straight.
I've seen a comment that the appearance depends on the model, but truly thats unhelpful, as often enough the customer is using a 3rd party model upon which they have no control of the geometry. In any case the exact reason its geometry-dependent isn't stated.
Here's a simple statement. As an ideal, No matter where the seam(s) are located they ought not to be visible. Let's not have a compromise approach, instead strive to the highest standard.

Jan-Soustruznik, 2024-09-17:
Hello, @bryn51, yes this thread is about the seam lines not being straight. We changed in 2.5.0 the seems placement algorithm which prefers regions which are not visible from the outside of the model at all or which are occluded from most directions, applied for seams set to Nearest or Aligned. New algorithm caused nonstraight lines on manifold objects. So we made lot of tests and enhancement the algorithm. Now the seams places are decided before slicing on model mesh, that cause improvement of problem about the placement with not affecting of the perimeter generator (classic/Arachne).

I think that the main topic about the visibiliti of the seems is created there - https://github.com/prusa3d/PrusaSlicer/issues/119142 about this issues continue there, please. (We did also there some enhancement in version 2.8.0, but the gap, missing material, etc depends on lot of Hardware factors, so for this topic for now doesn't exist any solution - so any idea is welcome to this tema).

thank you, have a nice day. 

> So this issue is about the seam lines not being straight ? But another report about the seam lines being too obvious closed in favour of this one?
>
> Seems (obvious pun) that whats needed is [...]

closed 2026-09-29 by prusaslicer-bot[bot]