prusa3d/PrusaSlicer #8446: Generating detail not on model

NotJannet, 2022-07-01:
### Description of the bug

2.5.0-alpha2 creates geometry not existing on the model.

[image]
[image]

It only in this orientation, rotating it produces expected results.

### Project file & How to reproduce

Included project file for this.
[files.zip](https://github.com/prusa3d/PrusaSlicer/files/9025114/files.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2

### Operating system

Windows 10 21H2

### Printer model

Original Prusa i3 MK3S+ MMU2S

hejllukas, 2022-07-01:
@NotJannet Thanks, reproduced. We will look at it.

hejllukas, 2022-07-26:
Thank you for reporting this issue. It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/77f5973c25ef0edc6db95de13c5976722b57a023 in master_250, and it will be part of the next alpha or beta.

closed 2022-07-26 by hejllukas