prusa3d/PrusaSlicer #8447: "Flow::spacing() produced negative spacing" error on 0.15mm print profile with 0.4 nozzle

kolba, 2022-07-01:
### Description of the bug

Whenever I try to slice the [attached model](https://github.com/prusa3d/PrusaSlicer/files/9029208/bagr.zip) on 0.15mm print profile, I get `Flow::spacing() produced negative spacing. Did you set some extrusion width too small?` error.
[image]

- The print profile is not changed. 
- Seem to error out on both 0.15mm QUALITY and SPEED. 
- Same result happens on Prusa MINI or MK3S printer with 0.4 nozzle. 
- Seem to be ok with smaller or bigger nozzle size. 
- Other STL models seem to slice fine. 
- Works in 2.4.2 but fails in 2.5.0-alpha2

### Project file & How to reproduce

1. Open [attached project](https://github.com/prusa3d/PrusaSlicer/files/9029208/bagr.zip) in prusaslicer 2.5.0-alpha2
2. Select Prusa mini or mk3s as a printer (default with 0.4 nozzle)
3. generic pla as filament (doesn't seem to matter)
4. 0.15mm QUALITY or SPEED print profile
5. click "slice now"
6. You should see the "_Flow::spacing() produced negative spacing_" error

[bagr.zip](https://github.com/prusa3d/PrusaSlicer/files/9029208/bagr.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2

### Operating system

Windows 11

### Printer model

Prusa MINI

closed 2022-07-11 by hejllukas

DrSpaltenstein, 2022-10-13:
Ive the same Problem in 2.5.0+ win64 202209060714 when I use Archne Perimeter generator.
[image]

I get Flow::spacing() produced negative spacing. Did you set some extrusion width too small? error.

chripopper, 2022-12-23:
I also get this error with 2.5.0 win64 and this model: https://www.printables.com/model/236318-vecna-001-stranger-things
It only occurs if i enable supports with any settings.

indiechixor, 2023-02-09:
> Ive the same Problem in 2.5.0+ win64 202209060714 when I use Archne Perimeter generator. [image]
> 
> I get Flow::spacing() produced negative spacing. Did you set some extrusion width too small? error.

Thank you. This issue is still occuring when attempting to slice model I created in Fusion360. 

I initially downloaded PrusaSlicer 2.5.0, saw this error existed in previous versions (this issue) and a fix would be included in the alpha. Downloaded 2.6.0 alpha3, still getting issue.

Found your comment and changed from Archne perimeter to classic, and resolved the problem. 

This does seem like a larger issue with this perimeter setting.

forReason, 2023-09-15:
Issue still exists in 2.6.1

Aerospacesmith, 2023-10-06:
Just encountered this in 2.6.1. Seems to be an issue with Arachne

FidelCapo, 2023-10-09:
@forReason and @Aerospacesmith Can you please send us the 3mf files in which you are getting this error? Thank you

Aerospacesmith, 2023-10-09:
@FidelCapo [04_bodyPart_d_FDM (repaired).zip](https://github.com/prusa3d/PrusaSlicer/files/12851101/04_bodyPart_d_FDM.repaired.zip)

robot256, 2024-01-18:
> I also get this error with 2.5.0 win64 and this model: https://www.printables.com/model/236318-vecna-001-stranger-things It only occurs if i enable supports with any settings.
> 
> I have the model rotated so its standing up, and scaled it to 170%. I'm trying to support the chin.
> 
> EDIT: Never mind. I've recently switched from super slicer to prusa slicer and thus exported/imported my profiles. Somehow max layer height limit (under printer settings) was set to 75mm instead of 75% of nozzle size. When i changed to 0.3mm it worked. So I guess you can try with a defalt profile to see if you have any error in yours.

Might not be related to the original but, but I just want to say THANK YOU.  I've been struggling for months with flow:spacing errors when switching between organic and snug supports on some of my profiles.  Turns out the Voron profile I was using also had max layer height set to 75mm, and lowering it to 0.4mm fixed the problem.

Wishingtreetops, 2024-02-01:
> I also get this error with 2.5.0 win64 and this model: https://www.printables.com/model/236318-vecna-001-stranger-things It only occurs if i enable supports with any settings.
>
> I have the model [...]

this was my exact issues, thank you for your comment. i was getting all sorts of confused and worked up. THANKS!!!

martinmakesit, 2024-08-04:
I ran into this issue recently and the fix for me was to change "Minimum wall width" under Wall Generator.  I was using a .3 nozzle with .3 walls and had to increase the value to 92.

yapalexei, 2024-11-01:
I also had a similar issue that occurred for me when I enabled multi-material printing. And what fixed it for me was setting the outer wall line width parameter to a none-zero value like "0.4 mm". My flavor of slicer, based on prusa, by default, had set all the line width values (except the default width) to zero. I suppose when a parameter value is 0 it will "default" to what the "default" value is. And this had worked as such until multi-material printing was enabled which seemed to have set the "perimeter_extrusion_width" to an actual 0.

However, after looking at the code that emits this exception/error, I think the message could, at least, be a bit more descriptive to help the user recover from their mistake by setting the "perimeter_extrusion_width" to a none-zero value.

Diegolgo, 2024-12-01:
hello, i have this problem. the solution is kinda simple (reading your and other comments...)
you need to change the widht manually in advanced, extrusion width, all values input manually for example 1.124xnozzle width (in my case 0.4), = 0,42  (this work with other values for support 0,35 etc etc)
i always use "0" (automatic) in all values, and only in some cases i changed.. (this is one of those)
there's other option you can try, but this gonna resolve this exact "problem" or bug.
[image]
[image]

just for data, this is the error:

[image]
you need fix this:
[image]

pd:  i use arachne