prusa3d/PrusaSlicer #8451: 2.5.0-alpha2: missing  >=boost-1.79.0 and >=cereal-1.3.1 support

slashbeast, 2022-07-02:
### Description of the bug

All the fixes for the interface change from boost::filesystem to boost::nowide around *fstreams are missing from 2.5.0-alpha2 even though they were long fixed on master branch.

I cannot find what branch has the code beside tag version_2.5.0-alpha2 so instead of merge request here you have patches:

- boost compilation failures - https://github.com/gentoo/gentoo/blob/master/media-gfx/prusaslicer/files/prusaslicer-2.5.0_alpha2-boost-fixes.patch
- cereal linking failure - https://github.com/gentoo/gentoo/blob/master/media-gfx/prusaslicer/files/prusaslicer-2.4.0-fix-build-with-cereal-1.3.1.patch

Once both are applied on the top of 2.5.0-alpha2, the PrusaSlicer can be built on modern Linux system with boost-1.79.0 and cereal-1.3.2.

### Project file & How to reproduce

Try to build PrusaSlicer on any updated rolling release distribution.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2

### Operating system

Linux, Gentoo, ~amd64

### Printer model

Prusa mk3s+

slashbeast, 2022-07-11:
Guys? All you have to do is apply 2 patches and commit them.

lukasmatena, 2022-07-12:
Thanks, but we will not backport the fix from master into 2.5. The 2.5 is a relatively small improvement over 2.4.2, it is already tested (with boost 1.75 and cereal 1.2.2) and we didn't and don't want to risk introducing some hard to find bug by applying the patch. Sorry.

closed 2022-07-12 by lukasmatena