prusa3d/PrusaSlicer #8452: Lightning infill reliably crashes 2.5-alpha2+x64 on macOS Monterey (21F79)

parhelia, 2022-07-02:
### Description of the bug

Running macOS Monterey, slicing any model with lightning infill guarantees a crash when slicing begins.

Steps to reproduce:
1. Import any model into the slicer
2. Go to Print Settings>Infill>Fill Pattern and choose "Lightning"
3. Slice

What I expected to happen:
The model should generate GCODE with lightning infill

What actually happened:
PrusaSlicer2.5-alpha2+x64 crashes 100% of the time

Workarounds:
None.

Things I've tried:
Rebooting and reinstalling PrusaSlicer 2.5-alpha2+x64

### Project file & How to reproduce

[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/9032844/Archive.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5-alpha2+x64

### Operating system

macOS 12.4 Monterey (21F79)

### Printer model

Prusa Mk3S+

bradley9999, 2022-07-02:
FWIW on MacOS 13 Ventura on a Mac Studio base model, lightning infill slices with no problems.

hejllukas, 2022-07-18:
Thank you for reporting this issue. I was able to reproduce it on 2.5.0-alpha2. In released [2.5.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha3), we fixed this issue.

closed 2022-07-18 by hejllukas