prusa3d/PrusaSlicer #8454: Profile Management Options

Zommy5re, 2022-07-03:
Hi, I'm a student in a secondary school and I manage a range of 3d printers at my school, including multiple Prusa Minis and a modified CR-10S PRO. I have to build and organise profiles in order to make it as easy as possible for other students less familiar with 3d printers to slice files, and I have run into the issue where students tend to just try use the default print profiles for all machines, which is an issue for the CR-10 as it is now running a duet board, direct drive and needs custom start gcode. Whilst I could try teaching everyone to only use the user presets I have built, the unfortunate reality is that most will just ignore this and use them anyway.

A really helpful (and hopefully easy to implement) solution would be to have an option in preferences where default profiles can be hidden if the user has their own profiles they'd prefer to use. Obviously default profiles shouldn't be deleted, so simply hiding them with an option to restore them would be best. Ideally these could be split into "Suppress System Print Presets", "Suppress System Material Presets", and "Suppress System Printer Presets"

I'm not familiar with coding, but simply doing it in a way similar to the ''Suppress default presets'' option would be best, simply instead suppressing the system ones. Would make sense to have this unchecked by default, but the ability to enable it would make people like me's jobs a lot easier, trying to make printers work for those who are less experienced. 

An alternative would be to have user presets ordered above the defaults, and allow the renaming of defaults. This way users would have their own profiles (which they are more likely to use) on top of the defaults, and could rename the defaults something like ''do not use'' or similar to dissuade other users from choosing them. Any other ideas that would help with making it easier to hide default settings that could lead to inferior prints or possibly damage a custom machine if less experienced users were to try slicing with them would be welcome too. 

Thanks for your consideration, 
Zac :)

dartrax, 2022-07-09:
May be this could be a workaround? #4538

stepikovo, 2022-07-10:
I also believe the workaround should work. Closing

closed 2022-07-10 by stepikovo