prusa3d/PrusaSlicer #8457: Pressure equalizer omits feedrate parameter on first print move after travel

crx-cooper, 2022-07-03:
### Description of the bug

Under some circumstances the new pressure equalizer appears to be printing at travel speed. So far I've only observed this in attached project, and **only on layer 54**. I'm a big fan of the new pressure equalizer and already had some great results otherwise.

[image]

### Project file & How to reproduce

1.Open [pressure_equalizer_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/9035282/pressure_equalizer_issue.zip)
2.Slice
3.Select 'Speed' view from the drop down bottom/left
4.View layer 54. Expected print speed would be between 15 and 40 mm/s but it's 120 mm/s on that layer.  When turning pressure equalizer off (0) the issue goes away. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha2+win64

### Operating system

Windows 10 21H1

### Printer model

Geeetech A20M with Marlin 2.0.7.2

lukasmatena, 2022-07-07:
Thanks, reproduced. We will look into it.

hejllukas, 2022-07-22:
Thank you for reporting this issue. It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/dd6096bf0b2c4c9ebadd97ec704eb7ddfaedeb2e in master_250, and it will be part of the next alpha or beta.

closed 2022-07-22 by hejllukas