prusa3d/PrusaSlicer #8458: Single-layer preview of vase mode STL does not show entire layer

timur-tabi, 2022-07-03:
### Description of the bug

Sliced my STL in vase mode.  In full preview mode, it looks okay.  If I show only some layers, it also looks okay.  But if if I show only one layer, then one of the four sides is not shown.

### Project file & How to reproduce

[vase_layer_preview_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/9035623/vase_layer_preview_bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2 and 2.4.2

### Operating system

Linux

### Printer model

MK3s

lukasmatena, 2022-07-07:
This is already fixed, the fix will be in 2.6.0 release.

kubispe1, 2022-12-14:
The issue is fixed in 2.6.0. I am closing it. Thank you for your comment.
[image]

closed 2022-12-14 by kubispe1