prusa3d/PrusaSlicer #8459: Tooltip on "Retract amount before wipe" setting for retractions

rcmaniac25, 2022-07-04:
PrusaSlicer 2.4.2 Windows 10 x64

Small background: I was getting some retraction settings for someone and ended up looking at what retraction settings existed. I saw "Retract amount before wipe", which was a percent and not a distance and wasn't entirely sure what it was. I hovered over the text and there was no tooltip.

After thinking about it a bit, I realized what it was ("what % of retraction needs to happen before it performs a wipe"). Closest I found was "This option does a quick retract before doing the wiping movement. More suitable for the Original Prusa MINI/MINI+." in a post about dealing with stringing.

In the end, it didn't seem as obvious as "retraction speed" and "extra length on restart" that also don't have tooltips. Which is why this ticket was created.

dartrax, 2022-07-06:
This is interesting. As fas as I understand there are three places where the tooltips should appear:

- Parameter label
- Parameter value (with additional information of default value and parameter name)
- Parameter unit

But some are missing indeed. For example, "Retract amount before wipe" misses the Label tooltip, the value tooltip does not appear when the field is disabled (this is by purpose I believe), only the unit tooltip is there. So you need (to know) to hover the "%" sign to get the tooltip.

[image]

lukasmatena, 2022-07-12:
Thanks for the report. The issue should be fixed by 1afc6fb57caa481a76d35dfe8001751ecb7b72f7, it will be ok in 2.5.0-alpha3. Closing.

closed 2022-07-12 by lukasmatena