prusa3d/PrusaSlicer #8460: Arachne engine fills small concentric gaps in object

FruitieX, 2022-07-05:
### Description of the bug

I tried printing some of [these clamps](https://www.printables.com/model/122320-one-for-all-cable-strap-print-in-place-clamp-belt) with PrusaSlicer Version 2.5.0-alpha2+win64, and the new Arachne engine would slice the mechanism in such a way that it doesn't work.

Arachne:
[image]

Classic:
[image]

### Project file & How to reproduce

[ClampV2.zip](https://github.com/prusa3d/PrusaSlicer/files/9043480/ClampV2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha2+win64

### Operating system

Windows 11

### Printer model

Original Prusa MINI+

neophyl, 2022-07-05:
Adjust your Slice Gap closing radius from 0.049 (I added an extra zero to 0.0049) and it wont join the islands.  Looks like arachne needs different settings due to the different algorithm.

FruitieX, 2022-07-06:
Thanks, dropping this down to 0.042 from 0.049 seems to be enough to fix this particular model. What downsides does reducing this setting come with - is this something that should be done by default to make more print in place designs work with the new slicing engine?

neophyl, 2022-07-06:
Well on all my personal profiles I have it set to 0.0049 by default.  Hasn't caused any issues in several years.  After all if the model has a gap there then it should have a gap when sliced.

Personally I think the default should be changed as it has caught out several people with tolerance models even with the 'old' slicing algorithm.  However PRUSA are very careful about changing defaults on profiles (rightly so).

hejllukas, 2022-07-14:
I looked at this issue, and it isn't directly an issue with Arachne. For this model `and Slice gap closing radius `equal to `0.049`, both classic and Arachne perimeter generators get this input:
[image]

You can see that the thin gap in the model has broken up into small pieces forming tiny holes. So given that input, Arachne returns the correct output, although it doesn't match the expected output. 
The reason why the classic perimeter generator returns the expected output is that it just got lucky. The algorithm of the classic perimeter generator unintentionally merged these small pieces together, and because of that, it unintentionally produced the expected output.

As @neophyl correctly mentioned, a smaller value for `Slice gap closing radius` solves this issue because the tiny gap no longer breaks into tiny holes. But in general, it doesn't mean that just Arachne needs a smaller value of `Slice gap closing radius`.

For completeness, there is the input to the perimeter generators when that value of `Slice gap closing radius` is set to `0.041`.
[image]

mix579, 2022-07-23:
I ran into the same issue. Good to know there's a workaround but I hope the default gets adjusted to the value suggested here.

kubispe1, 2022-09-15:
@hejllukas thanks for explanation. I label it as Arachne related issue to keep it in tasklist.

closed 2026-09-29 by prusaslicer-bot[bot]