prusa3d/PrusaSlicer #8461: Hang on slicing - brim creation 88%

sturze, 2022-07-05:
### Description of the bug

Hi, 
I ran into an issue which I can easily reproduce.
Slicing the model with 30% infill, 5 perimeters, and brim makes prusaslicer hang on brim creation (88%)
[image]

### Project file & How to reproduce

[tilecutterfreeze.zip](https://github.com/prusa3d/PrusaSlicer/files/9044538/tilecutterfreeze.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2

### Operating system

macOS 12.4

### Printer model

Mk3s+ with MMU2S

JohnOCFII, 2022-07-05:
If I uncheck the `Brim` selection, PrusaSlicer crashes during slicing (instead of hanging).  Changing to rectlinear infill did not change the hang on my Arm Mac.

New Gist: https://gist.github.com/JohnOCFII/690d8275fcd65c7e322a1f475ccc1b5c

plug-it-in, 2022-07-30:
I have this issue on 2.4.2 and 2.5 Alpha

PaulKC, 2022-12-08:
This is still happening for me in 2.5.0

hermetheuscoffee, 2023-01-14:
I don't know if this is bad form to reply to a closed topic, but this exact issue still exists for me on Win10 OS with PrusaSlicer 2.5.0. When slicing a model with a brim, it freezes at 88%. The only way to stop it is to forcefully kill PrusaSlicer. If I don't enable the brim, it slices just fine. Can this be re-opened?

PeterJacobsenDenmark, 2023-03-06:
Same issue for me in version 2.5.0

lukasmatena, 2023-03-07:
Does it also happen in [2.6.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5) ?

JohnOCFII, 2023-03-07:
>Does it also happen in [2.6.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5) ?

The original poster's example file (the tile cutter)  (which froze for me as explained above previous to 2.5) works fine in 2.6-alpha5.  I think the recent poster must have a separate, but perhaps related issue.

lukasmatena, 2023-03-07:
@JohnOCFII Thanks. In that case, we would like @hermetheuscoffee to send us the 3MF project, preferably into a new issue.

sturze, 2023-07-08:
sorry... somehow missed out on it... 
seems to be fixed in 2.6!

lukasmatena, 2023-07-10:
Thanks. I believe we can close this then. If anyone has similar problems, please, report it in a separate issue.

closed 2023-07-10 by lukasmatena