prusa3d/PrusaSlicer #8463: Crash during "Generating perimeters" with Arachne

jkavalik, 2022-07-05:
### Description of the bug

The slicer crashes silently during generation of perimeters when Arachne is enabled, works fine when switched back to Classic.
Loading the file is ok, then when slicing it hangs for some (varying) time at "Generating perimeters (20%)" and then the window closes, no error message, nothing to be seen when run using the console exe.

PrusaSlicer
Version:   2.5.0-alpha2+win64
Build:     PrusaSlicer-2.5.0-alpha2+win64-202206281406

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 16,914MB
OpenGL installation
GL version:   4.6.0 NVIDIA 512.78
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3050 Ti Laptop GPU/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
.. removed ..
</details>

### Project file & How to reproduce

Encountered when trying a clearance check test print from https://www.printables.com/model/160112-way-over-engineered-pull-copter-v3/ - attaching my copy of the original [clearance_check.zip](https://github.com/prusa3d/PrusaSlicer/files/9047899/clearance_check.zip) and my project file before slicing [clearance_check_problem.zip](https://github.com/prusa3d/PrusaSlicer/files/9047938/clearance_check_problem.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2+win64-202206281406

### Operating system

Windows 10 (build 19044), 64-bit edition

### Printer model

MK3S, Mini

jmd73, 2022-07-06:
Reproduced with 2.5.0-alpha2 on Windows 11 using clearance_check_problem.zip.

After a repair by the Netfabb Service, or splitting object to parts/objects, it slices without crashing.
Repaired project attached. Only opened @jkavalik supplied zip, repaired by netfabb, saved as project and finally zipped the file.
[clearance_check_problem_repaired.zip](https://github.com/prusa3d/PrusaSlicer/files/9052278/clearance_check_problem_repaired.zip)

jkavalik, 2022-07-06:
Did some more tests and almost everything fixes it - separating the file to parts, importing and exporting through Meshmixer (without any actual changes) etc.
So my guess would be the geared shaft part, where the top cylinder is placed on top of the geared part as a separate shell and because of some floating point magic ended in a state of intersections (in this version of the file) that breaks something.

hejllukas, 2022-07-12:
Thank you for reporting this issue. It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/e86463a369183134df95d5bc7d7c8a6387117edf in master_250, and it will be part of the next alpha.

closed 2022-07-12 by hejllukas