prusa3d/PrusaSlicer #8464: 2.5.0-alpha2 Aligned Seam placement issue

oozebot, 2022-07-05:
### Description of the bug

In 2.5.0-alpha2, aligned seam placement seems much more hellbent on doing its own thing instead of what is asked. Here is an example - I don't want the seam to be in the location that it's randomly generated, so I exclude that area. The expectation is that the seam be moved away from this area as it worked in 2.4.

Excluding an area that I do not want the seam.
[image]

Resulting seam does not follow this exclusion.
[image]

Another example is if I want the seam to be enforced to a corner. The expectation is that the seam will properly align itself to the full height of the enforced area as it sort of worked (with some coaxing) in 2.4.

Enforcing seams to align in a corner.
[image]

Resulting seam does not follow this enforcement.
[image]

### Project file & How to reproduce

Load most any functional model with square corners. Exclude areas for the seam to be excluded based on the initial seam placement. Slice model and notice the Aligned seam position does not follow this exclusion. Same for Enforcement.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2

### Operating system

Windows 11

### Printer model

oozeBot Elevate

Jan-Soustruznik, 2022-07-07:
Hello, oozebot, 

could you share a .stl file, where is the problem located. 

Thank you

oozebot, 2022-07-07:
The attached zip files contains a 3mf that demonstrates the issue. One instance of the object has seam enforcement, the other has exclusion. Neither are being followed. Perhaps it has to do with the new algorithm to hide seams inside.

[PrusaSlicer seam test.zip](https://github.com/prusa3d/PrusaSlicer/files/9063087/PrusaSlicer.seam.test.zip)

oozebot, 2022-07-08:
Another example where screw holes throw it off. It is much more desirable to fully align the seams and hide only the additional required seams.

[PrusaSlicer seam test2.zip](https://github.com/prusa3d/PrusaSlicer/files/9071782/PrusaSlicer.seam.test2.zip)

kubispe1, 2022-09-07:
This issue is fixed in PrusaSlicer 2.5.0-rc2, so I am closing it.

closed 2022-09-07 by kubispe1