prusa3d/PrusaSlicer #8467: G-code Viewer do not render "Tools" changes (in v2.5.0 or v2.4.? but was ok in v2.3.2 ...)

PPAC37, 2022-07-06:
### Description of the bug

from v2.4.?, the same .gcode (with a dual extrusion ( T0 , T1 )  ) do not rendrer tools changes.

in v2.3.2 (ok, as you can see two diffrent tools)
[image]

but in v2.5.0-alpha2 and other v2.4.x i try ( not ok only one tool ... but this is the same .gcode file )
[image]

( maybe related to #7548  and or #8324  but I speak without knowing. )

### Project file & How to reproduce

.gcode file used [dual.zip](https://github.com/prusa3d/PrusaSlicer/files/9059250/dual.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2+linux-x64-GTK3

### Operating system

Ubuntu 20.04

### Printer model

K8400

Kachidoki2807, 2022-07-07:
Hi,
Reproduced under win64 using v2.3.0 (ok), and v2.4.2 (nok) and 2.5.0-alpha2 (nok).

I can also see weird things about the temperature, which seems more or less linked to this issue.
[image]

For instance, the current tool is T0, and M104 T1 S190 apply the new temperature setpoint to this tool (green part) which is wrong,, followed by M104 S185 (blue part) which apply to T0 too which is correct (no tool specified). Those M104 are probably for pre-heating T1 and cooldown T0 before the upcoming toolchange.

I suspect a mistake during the parsing of 'T'.

Thanks @PPAC37 for reporting this issue.

lukasmatena, 2022-07-08:
Thanks for the report. it seems we really broke loading of external G-Codes. This G-Code is from Cura and the `T1` toolchanges are currently dismissed as invalid in `GCodeProcessor::process_T`, because `m_result.extruders_count` is empty. I do not know when it broke, maybe in https://github.com/prusa3d/PrusaSlicer/commit/b3ca8cb0af264d7e7ea285de03978b5185c3ff14 or a similar change where the way of determining the number of used extruders changed. @enricoturri1966 will fix it.

Kachidoki2807, 2022-07-09:
I had a look in `GCodeProcessor::process_M104`:
```cpp
void GCodeProcessor::process_M104(const GCodeReader::GCodeLine& line)
{
    float new_temp;
    if (line.has_value('S', new_temp))
        m_extruder_temps[m_extruder_id] = new_temp;
}
```
'T' parameter is not processed. It is purely ignored, and the temperature is applied to the current `m_extruder_id`. That explains the issue regarding the temperature.

It is unrelated to the OP issue (which is the result of the empty `m_result.extruders_count`), sorry for my bad assumption, even though this was really a second issue.

kubispe1, 2022-12-13:
In 2.6.0 G-Code preview is fixed also parsing of G-Code M104 works. Tool should render by now and temperature should be also fine. I am closing the issue.
[image]
[image]

closed 2022-12-13 by kubispe1