prusa3d/PrusaSlicer #8468: No smooth arcs

sanczess, 2022-07-07:
### Description of the bug

[image]
[image]
[image]
Streight lines and small arcs are no problem. Wide arcs are stuttering which results in uploaded pictures.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 alpha

### Operating system

Windows 10

### Printer model

Flsun q5

rtyr, 2022-07-07:
Please provide your project file and gcode.

What makes you think that stuttering is related to PrusaSlicer?

rtyr, 2023-09-12:
No response. I am closing the issue.

closed 2023-09-12 by rtyr