prusa3d/PrusaSlicer #8470: re-import of more-part stl

Ninister, 2022-07-07:
**Is your feature request related to a problem? Please describe.**
when I load an stl-file with more than one part, I usually split the parts and arrange them on the print area, some parts gets rotated, that they lay on the bed. after some correction in my modeling software I can't re-import the whole stl. I have to delete all the parts and to start the import from the beginning 

**Describe the solution you'd like**
re-import more parts stl like one part stl

**Describe how it would work**
the stl file have some hierarchy or IDs, so its possible to indentify the single parts over and over again. the software can compare the parts based on polygon count or size of the objects or the positions at the start. even if its not possible, because the new stl maybe have some additional parts or because the comparing gets error, the user can manually assign the parts to the ones on the print bed.

**Describe alternatives you've considered**
manually assignment or by identical name of the parts

**Additional context**
Add any other context or paint drawing about the feature request here.

Ninister, 2022-07-07:
[image]
[image]

TomaszSzt, 2022-07-11:
The re-import (F5) should also apply to files imported by "split to objects". 

Alternatively the warning "Cannot re-load split parts" should be displayed to inform user about the fact, that parts did not change even tough it was requested.

closed 2026-09-29 by prusaslicer-bot[bot]