prusa3d/PrusaSlicer #8472: Arachne inner Perimeter missing when Perimeter distribution count = 3

dartrax, 2022-07-07:
### Description of the bug

I noticed a missing single perimeter path. I changed "Perimeter distribution count" to 3, it doesn't occur when I use other values.
I wasn't sure if it's a bug or intended behavior until I noticed it doesn't happen when I cut the model in CAD (leaving the rest of the geometry untouched). Also I can't fill this space with Gapfill.

[image]

### Project file & How to reproduce

[Arachne-Test.zip](https://github.com/prusa3d/PrusaSlicer/files/9065974/Arachne-Test.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha2+win64

### Operating system

Win 10 64bit

### Printer model

Artillery Sidewinder X1

jmd73, 2022-07-07:
This is a weird one. After fixing the object with the Netfabb service it slices fine.
Even just exporting the object as STL and loading as a new object, it also slices fine.

Image of the repaired object:
[image]

dartrax, 2022-07-17:
Also present in Alpha 3.
I also wanted to note that I could not find any mesh issues in the model. It was generated with Autodesk Inventor and I've never had any mesh issues with exports from that cad.

hejllukas, 2022-08-02:
Thank you for reporting this issue. It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/0161a59a93e326d5e573cffe61b9549a4d925189 in master_250, and it will be part of the next alpha.

closed 2022-08-02 by hejllukas

dartrax, 2022-09-18:
I think this issue is not fully fixed:

![image](https://user-images.githubusercontent.com/15105943/19

[Hosenträgerverkürzer.zip](https://github.com/prusa3d/PrusaSlicer/files/9593746/Hosentragerverkurzer.zip)
0897314-2b9e674f-0ebc-48a1-86af-1c462a0b641a.png)