prusa3d/PrusaSlicer #8474: PrusaSlicer 2.5.0 Alpha2 GTK3 crashes on a certain stl file when selecting 0.05mm or 0.07mm layer height and clicking Slice now.

juckendesAuge, 2022-07-08:
### Description of the bug

Hello,
the crash is only on 0.05mm and 0.07mm Layer high.
PrusaSlicer-2.5.0-alpha2+linux-x64-GTK3-202206281406.AppImage crash after click Slice now
[Vic1Atom.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9068451/Vic1Atom.3mf.zip)

STL: Vic1Atom.stl on 0.7mm Layer high
https://www.thingiverse.com/thing:2839310/files

Terminal:
[juck@fedora PrusaSlicer]$ ./PrusaSlicer-2.5.0-alpha2+linux-x64-GTK3-202206281406.AppImage 
[2022-07-08 04:27:28.145434] [0x00007ff284cd2880] [trace]   Initializing StaticPrintConfigs
Speicherzugriffsfehler (Speicherabzug geschrieben)

### Project file & How to reproduce

Unzip and open the 3mf Project in the Attatchment and click Slice now.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha2+linux-x64-GTK3

### Operating system

Fedora release 35 (Thirty Five)

### Printer model

Prusa MK3s+

lukasmatena, 2022-07-08:
Thanks, reproduced. The issue seems to only appear when Arachne is active. We will look into it.

lukasmatena, 2022-07-08:
Btw. fixing the model by NetFabb apparently helps.

juckendesAuge, 2022-07-08:
Thanks for the info about Netfabb. I was able to install the Basic version.
Unfortunately PrusaSlicer crashes at 0.07mm layer height even after repairing the STL file.
Exactly because of Arachne I wanted to print it again. Since my printout was not so good at that time. If you select 0.2mm layer height, there is a bridge at layer 0.4mm. Therefore I wanted to go down with the layer height.

kolba, 2022-07-15:
> 

this worked for me

[image]

juckendesAuge, 2022-07-16:
Thank you @kolba 

i haven`t this in the options.

[image]

kolba, 2022-07-18:
ah, my bad, missed the whole linux setup thing. well, for what its worth, it happens in windows 11 as well as in fedora :). 2.5.0 alpha2 and alpha3

hejllukas, 2022-07-21:
Thank you for reporting this issue. It helped us discover a weird issue of the Voronoi diagram generator library. With commit https://github.com/prusa3d/PrusaSlicer/commit/63c66f4f189b6f3d3a3b95ddd158c66ee5166420, we introduced the workaround for those types of issues. So in master_250, it is fixed, and this fix will be part of the next alpha or beta.

closed 2022-07-21 by hejllukas