prusa3d/PrusaSlicer #8475: Internal error: Could not load bitmap: fuzzy_skin

Prop3r, 2022-07-08:
### Description of the bug

The program produces an error and closes on almost every action, and every model. 

Internal error: Could not load bitmap: fuzzy_skin

07:48:43: can't open file 'D:\_Prive\___3D Printer\Prusa slicer\PrusaSlicer-2.5.0-alpha2+win64-202206281406\resources\icons\fuzzy_skin.png' (error 2: Het systeem kan het opgegeven bestand niet vinden.)
07:48:43: Failed to load image from file "D:\_Prive\___3D Printer\Prusa slicer\PrusaSlicer-2.5.0-alpha2+win64-202206281406\resources\icons\fuzzy_skin.png".
07:48:43: Internal error: Could not load bitmap: fuzzy_skin

Strange thing is, it's searching for fuzzy_skin.png, but only fuzzy_skin.svg is present in the folder. Typo in the code? 

### Project file & How to reproduce

[FussQ.zip](https://github.com/prusa3d/PrusaSlicer/files/9069298/FussQ.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.5.0-alpha2

### Operating system

W10 64 bit

### Printer model

Artillery X1

lukasmatena, 2022-07-08:
@Prop3r Thanks for the report. It is quite strange that you are the only person suffering from this. Can you please download the 2.5.0-alpha2 zip again, unpack it into a separate directory and verify that you get the same error when running the binary in that directory? I remember a similar issue caused by downloading the zip and copying only the binary into the old installation.

> Strange thing is, it's searching for fuzzy_skin.png, but only fuzzy_skin.svg is present in the folder. Typo in the code?

Probably not. The way it currently works is that "fuzzy_skin" icon is requested, and the application tries to load an SVG. Only then it tries the PNG. So the right question is not "why it looks for a PNG", but instead "why it failed to load the SVG before that".

Prop3r, 2022-07-08:
Hello Lukáš,

I followed your instructions, downloaded alpha2 again, and installed on the same drive, but other folder location. This time the error did not happen, even after extensive use using all kinds of modelling tools and slicing options. Working stable now!

Thanks for the quick response, and the amazing new thinwall features of 2.5!

closed 2022-07-08 by Prop3r