prusa3d/PrusaSlicer #8477: Brim painting

altuor, 2022-07-08:
It would be great if brim could be 'painted, like for support or seams, I hope it's not a duplicate, I didn't find any similar request in the forum 

**Is your feature request related to a problem? Please describe.**
 I had a recent print where there was just a tiny part that would have needed brim to hold on the plate, it was just after 3hours print that it fell off, ruining the print. 

**Describe the solution you'd like**
A 'painting tool' that would allow to paint on 'edges' where brim is needed

**Describe how it would work**
The painted edges would generate brim

**Describe alternatives you've considered**
?

neophyl, 2022-07-08:
This issue already exists https://github.com/prusa3d/PrusaSlicer/issues/8283  please dont create duplicates

altuor, 2022-07-08:
Oh, didn't find that one, sorry :-o

lukasmatena, 2022-07-08:
Closing as a duplicate.

closed 2022-07-08 by lukasmatena