prusa3d/PrusaSlicer #8479: Exclude properties of the intro line from the legend

stefan-am, 2022-07-08:
**Is your feature request related to a problem? Please describe.**
The intro line has in most cases the highest values for width and volumetric flow rate (much higher than the print itself). 
The legend displays these values. 
The downside is, that many colors of the legend are “wasted” and only few are used for values of the real print.  

**Describe the solution you'd like**
I do not know who would be interested about properties of the intro line, so in my opinion it would make sense to exclude the properties of the intro line from the legend. 

**Describe how it would work**
The advantage would be, that more different colors would be available for a clearer display of the properties of the real print.

**Describe alternatives you've considered**
Maybe use a checkbox to display / hide the intro line and include / exclude the intro line properties into the legend.  

[image]

dartrax, 2022-07-09:
From what I understand of #7066 this should have been fixed in PrusaSlicer 2.4.0-beta2.
However, it may be possible to deactivate the visibility of the purge line, and there is already a FR asking if that can be a configurable default option: #7434
Is that what you need?

stefan-am, 2022-07-09:
#7066 is exactly what i described above. 
It was not obvious for me, that the "purge line" (in the G-code comments it is called "intro line") falls into the category "Custom". 
Maybe it would make sense to change the Checkbox name to "Custom/Prime Line" or "Custom/Intro Line" ?
For other people wondering i add this picture: 
The checkbox for feature type "Custom" controls the visibility of the purge line and the legend adopts to this selection :-) 
This issue can be closed from my side. Thank you! 

[image]

closed 2022-07-12 by stepikovo

bszonye, 2022-08-25:
The fix in #7066 is not working for me for Width in the legend. The range of widths includes the intro line even if I deselect Custom from Feature Types. I'm using 2.4.2 but there's also a report of this affecting previews in 2.5.0 where it makes it hard to see the results of Arachne output: https://forum.prusa3d.com/forum/prusaslicer/how-to-disable-viewing-of-custom-g-code-messes-up-width-preview-with-arachne-in-2-5/

bszonye, 2022-08-25:
I took a look at the commit that fixes #7066: https://github.com/prusa3d/PrusaSlicer/commit/e41928514748e1513584d7c058f6134269f91d96

That code only appears to affect Volumetric Rate, not Width, which would explain my findings above. It appears that this issue is only partly fixed. The intro line still affects other parts of the legend, notably Width which is much more relevant in 2.5 with Arachne.

RandomDude4, 2022-08-28:
I can confirm that I also have the issue in 2.5 RC1 so the purge line still affects the scale/values of the "Width" in the legend.

FidelCapo, 2023-04-24:
@bszonye Your issue should be fixed by adding new button in legend and fix will be part of next release.
Closing.