prusa3d/PrusaSlicer #8487: Print order change on PS 2.5

joseaguardia, 2022-07-11:
### Description of the bug

In previous versions when there are several objects, priority is given to start from the last in the list, but in version 2.5a it is the opposite, starting from the first in the list.

### Project file & How to reproduce

Add multiple STL files to the list and reorder them.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0-alpha2+linux-x64-GTK3-202206281406

### Operating system

Ubuntu 21.04

### Printer model

Anycubic i3 Mega S

lukasmatena, 2022-07-12:
The objects are printed in the same order as they appear in the object list ONLY when sequential printing is active. When it is not active, there is no guarantee and the order is chosen so that the travels are minimized (the algorithm might surely be improved, but it tries to minimize them). It may or may not print in the same order as in the object list.

Did you just notice it or is that a problem for you? If so, why?

psyonara, 2022-09-03:
I have the same issue with the ordering. When I do a manual colour change in a print with multiple objects, I add a "dummy" cylinder that is meant to prime the nozzle before the new colour starts printing. Not being able to set a specific print order makes it a bit annoying to try to get the cylinder to print first.

smirgol, 2022-11-22:
I have a similar issue, but with adding instances of an object:

I have 2 instances of an object. 
With 2.4 instance 1 is in the front, instance 2 in the back.
With 2.5, instance 1 is in the back and instance 2 in the front.

I can not reorder the instances manually.

That likely will cause issues with sequential printing, as it now starts with the one in the back, then moves forward to instance 2. Thing is, on the back of the extruder there are cables and it will crash into instance 1 for guarantee.
If it prints starting with the object in front, like it did with 2.4, this cannot happen.

Any ideas?

closed 2026-09-29 by prusaslicer-bot[bot]