prusa3d/PrusaSlicer #8488: Slicing hangs with certain ligthning infill percentages

pylorak, 2022-07-12:
### Description of the bug

With some infill percentages and lightning infill, slicing hangs indefinitely at "Infilling layers" stage.

Some examples for infill values that work correctly:
20%
30%
40%
60%
80%

Some examples for infill values that DO NOT work:
50%
52%
67%
70%

### Project file & How to reproduce

1. Open attached project file ([Case_V1_bugreport.zip](https://github.com/prusa3d/PrusaSlicer/files/9088367/Case_V1_bugreport.zip)).
2. Click "Slice now".
3. Result: Slicing hangs.

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0-alpha2

### Operating system

Windows 10

### Printer model

Voxelab Aquila X2

lukasmatena, 2022-07-12:
@pylorak Thanks, reproduced. In Debug mode it crashes on `assert(self.m_unsupported_points_bbox.contains(sp));` in `DistanceField` constructor. Fixing the model by NetFabb helps.

hejllukas, 2022-07-12:
Thank you for reporting this issue. It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/22255822a22399b15d1096868a9e2ae9d64f4e9a in master_250, and it will be part of the next alpha.

closed 2022-07-12 by hejllukas