prusa3d/PrusaSlicer #8490: Print Head Crashes into print bed when print completes

Thybigwheel, 2022-07-12:
### Description of the bug

When the print completes, the print head goes all the way up and then comes all the way down to the Z zero point and extends to the maximum of the X-axis. It ignores any printed material on the plate and crashes the head into the printed part. I cannot find anything in the manual or the slicer program on how to stop it form doing this.
[image]
[image]
[image]
[image]
[image]
 

### Project file & How to reproduce

[Assembly fixtures.zip](https://github.com/prusa3d/PrusaSlicer/files/9094717/Assembly.fixtures.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+win64

### Operating system

Windows 10

### Printer model

Original Prusa MINI

neophyl, 2022-07-12:
What version of mini firmware are you using ?  There’s a known fixed issue with the head moving down that was fixed in 4.3.2 
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.2  So I’d suggest updating to the latest if you aren’t already and see if the issue persists.

stepikovo, 2022-07-12:
As neophyl mentioned, it's an FW issue, please update your printer :)

Thybigwheel, 2022-07-13:
Aw shoot! Thanks Guys!!