prusa3d/PrusaSlicer #8491: PrusaSlicer non-alpha somehow opening when using "Open with..." dialogue to select the PrusaSlicer alpha build directly.

ggppjj, 2022-07-12:
### Description of the bug

With the stable version of PrusaSlicer being installed via the experimental installer from Prusa's download page, I am unable to have Windows open an STL with a downloaded alpha PrusaSlicer release by using "Open with..." as selecting the alpha PrusaSlicer executable in the dialog still only ever opens the file in the non-alpha build.

I've attempted deleting the .stl file association in my registry which did require me to pick a program to open STLs with. Picking the alpha PrusaSlicer build causes Windows to open the file in the non-alpha version that is installed.

I don't currently think that this is an issue with Windows, although it wouldn't surprise me to find out that it is. I currently have to open PrusaSlicer Alpha and pull files in once it's open, but this is personally bugging me as to why I can't manually tell Windows what version of PrusaSlicer to open STLs with by default.

### Project file & How to reproduce

Reproduce:

Install PrusaSlicer via the experimental installer on the website.
Download and unzip the current alpha build.
Open an STL with non-alpha PrusaSlicer, close PrusaSlicer.
Right click the STL, use Windows's dialogue to "Open with" -> "Choose another app"
In the program selector, choose "Look for another app on this PC".
Find and select the PrusaSlicer Alpha executable.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2+win64, 2.5.0-alpha2+win64

### Operating system

Windows 11 Pro

### Printer model

Prusa MK3S+ MMU2S+

ggppjj, 2022-07-12:
Willing to accept that this is a Windows issue, but I have no clue how it could be. Other programs work correctly with the dialog, this does affect multiple of my computers.

stepikovo, 2022-07-17:
I'm closing this as a duplicate of #8332

closed 2022-07-17 by stepikovo