prusa3d/PrusaSlicer #8498: Feature Request: Temperature change on Filament Change and/or Layer Height

MrTantrum, 2022-07-13:
On my MK3S+ I print mostly multi-colored models (non-MMU) with one or more filament changes during the print.  While I always use the same filament type in my prints (e.g. all PLA), I do utilize different filament brands and colors.  Often times, these filaments have ideal print temperatures that may vary as much as 30 degrees.  

Currently, I have to remember to manually change the print head temperature manually upon filament change, but it would be nice if there was a feature built into Prusa Slicer where the print head temp could be set in accordance with a filament change (preferred) and/or the temp could be set at a given layer height.

rtyr, 2022-07-13:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2002, https://github.com/prusa3d/PrusaSlicer/issues/2255, https://github.com/prusa3d/PrusaSlicer/issues/3439, https://github.com/prusa3d/PrusaSlicer/issues/3439.

As a workaround, you can add M104 Sxxx (xxx = desired temperature) custom command using vertical slider, so it will change the temperature automatically at the specified layer.

I am closing this as a duplicate.

MrTantrum, 2022-07-13:
Thank you for your response.  I understand how to inject G-code manually, but this is performed on such a frequent basis that it is quite inconvenient.  Would be extremely helpful if such a feature was embedded into the Prusa Slicer GUI somehow.

MrTantrum, 2022-07-13:
Actually, I just saw your comment about inserting the G-Code via the vertical slider.  I wasn't aware of this feature and serves as a valid and practical way to achieve what I'm requesting.  Thank you.

rtyr, 2022-07-13:
Either that, or you can only add custom gcode like this (2 lines - M600 command is for color change) 

```
M600
M104 Sxxx
```

MrTantrum, 2022-07-13:
Thank you for the suggestion.