prusa3d/PrusaSlicer #8503: Sliced model is unprintable due to empty layer gaps. 3D model is solid.

JasonMeyerDesign, 2022-07-14:
### Description of the bug

After slicing there are several layers missing making the model unprintable. Changing the orientation effects where/if the gap occurs. In one orientation the issue does not appear, but it is not printable in this direction. the 3D model/STL does not appear to have the gaps or any other issues.

### Project file & How to reproduce

Open project file and look at the layer view.
The project has the same part loaded  in three different orientations to show how the error changes.
You will see there is a gap in two of the orientations, but not the third, showing that the model is not the issue.
[Slicing_Error.zip](https://github.com/prusa3d/PrusaSlicer/files/9115877/Slicing_Error.zip)
.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2  (same issue with 2.5.0)

### Operating system

Windows 10 Pro

### Printer model

Prusa Mk3S

foreachthing, 2022-07-15:
Your Model may be solid, but surely are not water tight.

Before "Fix through Netfabb":
[image]

After "Fix through Netfabb":
[image]

JasonMeyerDesign, 2022-07-18:
forecasting was correct. The model did have a hole in it. It was just odd that it didn't show up in the one orientation.

closed 2022-07-18 by JasonMeyerDesign