prusa3d/PrusaSlicer #8504: "Aligned seam" does not generate aligned steams

pylorak, 2022-07-15:
### Description of the bug

The documentation [here](https://help.prusa3d.com/article/layers-and-perimeters_1748#seam-position) states that the "Aligned" option should arrange seams into lines. However, this does not work reliably at all. See attached picture and [project file](https://github.com/prusa3d/PrusaSlicer/files/9116728/aligned_seam_error.zip).

At first glance this issue might be related to #8464. However, unlike #8464, this issue is merely about arranging seams into continuous lines and has nothing to do with finding edges or constraining seams into specific regions of the model.

[image]

### Project file & How to reproduce

Open attached project file, slice the model, and have PrusaSlicer show seams.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0-alpha2

### Operating system

Windows 10

### Printer model

Voxelab Aquila X2

pylorak, 2022-07-15:
Of course, here it is.
[leg_back_right.zip](https://github.com/prusa3d/PrusaSlicer/files/9120500/leg_back_right.zip)

westranm, 2022-07-19:
I can confirm the seam position feature is broken in 2.5.0 Alpha 3. For attached file, I get expected results in 2.4.2 but 2.5.0 A3 does not provide expected results when choosing each of the different seem position options
[spoolRollerV2Frame.zip](https://github.com/prusa3d/PrusaSlicer/files/9143291/spoolRollerV2Frame.zip)

westranm, 2022-07-19:
Here are screenshots of rear seam position comparing output of 2.5.0 alpha 3 (Arachne off and on) and 2.4.2. Only 2.4.2 gets it right. 
[image]
![2 5 0A3 Rear Seam, Arachne 22071912](https://user-images.githubusercontent.com/40043138/179838599-f6c61b6a-86b8-439e-a8e9
[image]
-449d145dc684.png)

varukasalt, 2022-07-23:
Having the exact same issue. Even modeled a long skinny cylinder to test it. Seam was either all over the place or very wiggly. Was trying this on a brand new printer and thought I had a belt loose or something it was so bad. Tried the same model with 2.4 and seam was positioned as expected

murk-sy, 2022-08-10:
Another example in 2.5.0-alpha3 (seems unchanged from alpha2) - this is a block with 2 cylinder negative volumes, but similarly shaped stls give the same results. This kind of shape and print orientation is pretty common for me.
[image]

[image]

murk-sy, 2022-08-15:
2.5.0-beta1 produces slightly better results

[image]

varukasalt, 2022-08-15:
I mean that's slightly better but it's definitely still not right.

kubispe1, 2022-09-07:
This issue is fixed in PrusaSlicer 2.5.0-rc2, so I am closing it. In case let mi know.

closed 2022-09-07 by kubispe1

murk-sy, 2022-09-07:
@kubispe1 
2.5.0-rc2, aligned setting, box+2x negative cylinder as before, once again an improvement, but not flawless

I can imagine it's practically impossible to make seam positioning work for every model, but stuff like this is more common for me

[image]
[image]

kubispe1, 2022-09-07:
After seam upgrade in last version 2.5.0 – rc2, examples in this issue looks much better, I think. Despite this, there are still some imperfection, that we are collecting at the moment. So @[murk-sy](https://github.com/murk-sy), can you please provide also your .3mf file, because I can not find it.

[image]

[image]

murk-sy, 2022-09-07:
@kubispe1 File attached. 
[PSissue-8504-Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/9506290/PSissue-8504-Shape-Box.zip)

This is sort of a very "basic" version of a lot of my printed parts: a block-like shape with horizontal holes, and generally visible on one side, with text on the invisible side as well. Example of an actual part:
[image]
[image]

DragonFinlay, 2022-09-16:
@murk-sy If the brush size is big enough to create overlaps between two possible areas for placing seams, it's difficult for the slicer to know where you want the seam to be. If you choose a smaller brush to avoid overlapping areas, it works fine:

[image]
[image]

For the part with the text, I agree, no seams should be placed on the text. We will be addressing that.

murk-sy, 2022-09-16:
Fair enough, I didn't use seam painting until Arachne so I never even tried making precise seam lines. As you showcased it, that looks basically perfect. I'll just take a bit more time with seam painting for cases like that.

Nice to hear text stuff will be getting some optimisations as well, interested to see how very small text will improve.