prusa3d/PrusaSlicer #8505: Blue Screen of Death When Using PrusaSlicer

jmarsh12, 2022-07-15:
### Description of the bug

I've been using PrusaSlicer for about 3 months now as I just recently bought a MK3S+, and I've been loving it. No issues for most of that time. However recently within the past couple weeks, I've been getting an error from PrusaSlicer and GCODE viewer that says "PrusaSlicer requires OpenGL 2.0 capable graphics driver to run correctly, while OpenGL version 1.1.0, render GDI Generic, vendor Microsoft Corporation was detected. You may need to update your graphics card driver." Then after a few moments, I would get a Windows blue screen with the error code: "VIDEO_DXGKRNL_FATAL_ERROR." I am using a Dell XPS 9560 laptop with an Intel Core i7 7700HQ and an NVIDIA GTX 1050 GPU. The problem originally arose while I was using PrusaSlicer version 2.4.2, and it was seemingly random. I updated both the CPU and GPU drivers, and the problem persisted. I did a clean Windows install thinking it was a Windows problem since the install was rather dated. No dice. I used NVIDIA Control Panel and the Windows settings to force PrusaSlicer to use my GPU instead of my CPU. Still no luck. I contacted support, and they suggested I try using the 2.5.0 alpha release of PrusaSlicer. I downloaded it and it seemed worked fine. However, a few days passed and I encountered the same error message and blue screen. I seem to have tamed the beast for now though, as a system reboot and driver reinstall fixes it when it occurs, and the issue only happens when I leave PrusaSlicer open and idle for more than about 10 minutes. Otherwise it works just fine. It may be corrupting the driver after being idle for so long, but I'm not sure. So I'm posting it here in hopes that either someone else has encountered similar issues, or perhaps it is an issue with PrusaSlicer itself, and maybe this will help in the quest to solve said issue. 

### Project file & How to reproduce

[config-20220714.zip](https://github.com/prusa3d/PrusaSlicer/files/9117142/config-20220714.zip)
This contains the config file for my slicer as well as the STL file I was working with when the problem occurred
[image]
[image]
.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha-win64

### Operating system

Windows 10

### Printer model

Prusa i3 MK3S+

murk-sy, 2022-07-15:
Post also information from Help>System info menu, that contains GPU related information that the devs will find helpful.

mattm55, 2022-07-29:
I am getting the exact same issue on a Dell with GTX1050 and 2.4.x thru 2.5 Alpha.  Works fine at first but leaving idle kills it. Tried the same driver updates and reinstall. Maybe Ill try an older NVidia driver? Following to see if there is a resolution.
Matt

Niflheim123, 2022-10-04:
I am also experiencing this on a DELL Latitude 5591. Prusa slicer 2.5.0

Niflheim123, 2022-10-04:
[bsod.zip](https://github.com/prusa3d/PrusaSlicer/files/9703054/bsod.zip)
attached a dmp file

DusekM98, 2023-04-08:
Crashing for me as well. Prusa slicer 2.5.2, Acer Aspire A515-51G

thew1se, 2023-09-18:
Same for me on Dell Latitude 7501.  

Work initially then causes blue screen of death.  

I think this began with 2.5 for me but can't be sure when.   Hoped it would fix with 2.6 but still an issue.  

Please let me know what you need to troubleshoot.  Super frustrated.

bubnikv, 2023-09-18:
I am sorry, but BSOD is an indication of a hardware or driver issue.
Closing.

closed 2023-09-18 by bubnikv

thew1se, 2023-09-18:
If it only happens with prusa slicer and no other software then it is probably not a hardware issue.

mattm55, 2023-09-20:
Well, I didn't update any drivers and the older versions of Prusa worked fine. The n ew versions crash in about 15 minutes of inactivity. So be it but not convinced its a driver issue nVidia.

murk-sy, 2023-09-20:
I didn't look into every laptop mentioned, but if you have an integrated *and* discrete GPU, try disabling one and then the other and see if the crash occurs the same way.

It's a long shot but it could be something related to discrete GPU handing off data to integrated (or vice versa) and failing.

I believe drivers also allow you to select which graphics processor to run specific applications on - but I don't have a laptop like that so I can't test myself.

thew1se, 2023-09-20:
> Well, I didn't update any drivers and the older versions of Prusa worked fine. The n ew versions crash in about 15 minutes of inactivity. So be it but not convinced its a driver issue nVidia.

I agree!  Didn't happen with older versions.

mattm55, 2023-10-13:
I am using the "console" version with no hangups. Just thought I'd pass that along. I haven't tried ver 2.6.1 yet without console version.
-Matt

thew1se, 2023-10-16:
POTENTIAL FIX:

I seemed to have fixed this problem.........On my Inspiron 7506 2n1.  I update the Intel(R) Iris(R) Xe MAX Graphic Driver using the Intel (Not Dell) software update, AND remove the Adapter from Device Manager, so it reinstall clean, the issue has gone away.  Cheers.