prusa3d/PrusaSlicer #8508: Prusaslicer bed textures missing

TheMrCaveman, 2022-07-15:
### Description of the bug

Prusaslicer should have bad textures.
Most of them are vissible for a blink then turn black.

### Project file & How to reproduce

open prusaslicer and select any prnte
[image]
r

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2, 2.5.0 alpha 2

### Operating system

windows 11

### Printer model

mk3s+

stepikovo, 2022-07-18:
Can you please try the method described here [#8417](https://github.com/prusa3d/PrusaSlicer/issues/8417#issuecomment-1164240369)?

TheMrCaveman, 2022-07-23:
I am on the latest amd gpu driver, have a 6950xt and the only way I can solve this is removing textures. Due note I am in the beta driver testing thing so might also pass this onto amd.

stepikovo, 2022-07-26:
@TheMrCaveman can you please try a different version of drivers? When did this start happening?
Can you send us the _System info_ from the Help menu?

Yornik, 2022-08-01:
same problem newest amd drivers 

system info is here 
`PrusaSlicer
Version:   2.5.0-alpha3+win64
Build:     PrusaSlicer-2.5.0-alpha3+win64-202207151230

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 22000), 64-bit edition
Total RAM size [MB]: 17,102MB
OpenGL installation
GL version:   4.6.0 Compatibility Profile Context 22.7.1.220725
Vendor:       ATI Technologies Inc.
Renderer:     Radeon RX 5500 XT
GLSL version: 4.60
<details>
<summary>Installed extensions:</summary>
GL_AMD_blend_minmax_factor
GL_AMD_bus_addressable_memory
GL_AMD_depth_clamp_separate
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_int16
GL_AMD_pinned_memory
GL_AMD_sample_positions
GL_AMD_shader_trinary_minmax
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
[... 174 lines cut ...]
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_NV_blend_square
GL_NV_copy_image
GL_NV_primitive_restart
GL_OES_EGL_image
GL_OES_draw_texture
GL_OVR_multiview
GL_OVR_multiview2
GL_OVR_multiview_multisampled_render_to_texture
WGL_EXT_swap_control
</details>
`
[image]

marcandrelevesque, 2022-08-01:
Same problem here. but the bed svg renders correctly with software render switch. It started to happen with AMD 22.7.1 version of Radeon drivers. To be noted, I have to other issues what so ever with any other software UI render.

PrusaSlicer
Version:   2.4.2+win64
Build:     PrusaSlicer-2.4.2+win64-202204251110

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 34,261MB
OpenGL installation
GL version:   3.0 Mesa 12.0.0-rc2
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 128 bits)
GLSL version: 1.30
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_APPLE_packed_pixels
GL_APPLE_vertex_array_object
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_stencil_export
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_fragment_shader
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_context_flush_control
GL_KHR_debug
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_MESA_ycbcr_texture
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
</details>

mklengel, 2022-08-06:
Same problem.

PrusaSlicer
Version:   2.4.2+win64
Build:     PrusaSlicer-2.4.2+win64-202204251110

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (Erzeugungsversion 22000), 64-bit Edition
Total RAM size [MB]: 51,435MB
OpenGL installation
GL version:   4.6.0 Compatibility Profile Context 22.7.1.220725
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon RX 6700 XT
GLSL version: 4.60
<details>
<summary>Installed extensions:</summary>
GL_AMD_blend_minmax_factor
GL_AMD_bus_addressable_memory
GL_AMD_depth_clamp_separate
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_int16
GL_AMD_pinned_memory
GL_AMD_sample_positions
GL_AMD_shader_trinary_minmax
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
[... 171 lines cut ...]
GL_EXT_texture_sRGB_RG8
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_NV_blend_square
GL_NV_copy_image
GL_NV_primitive_restart
GL_OES_EGL_image
GL_OES_draw_texture
GL_OVR_multiview
GL_OVR_multiview2
GL_OVR_multiview_multisampled_render_to_texture
WGL_EXT_swap_control
</details>

3Delights, 2022-08-15:
I have just started to get this problem.  It happens in v2.4.x and 2.5.x Unfortunately I updated my AMD 5700X Win10 64 drivers and the PrusaSlicer configuration on the same day so cannot be sure which caused this error.  I deleted the config files from under AppData/Roaming and let PS create new ones but it didn't resolve the problem.

When PS loads the correct texture is shown for a second or 2 then it turns black just as in TheMrCaveman's screenshot.

PrusaSlicer 2.5.0 Beta 1
Windows 10 Pro x64
AMD 5700 XT
AMD Driver v22.7.1

PS System Info attached:
[SysInfo.txt](https://github.com/prusa3d/PrusaSlicer/files/9338876/SysInfo.txt)

mklengel, 2022-08-15:
I also updated to AMD Driver 22.8.1 and tried PrusaSlicer Version 2.5.0-beta1+win64.
No luck = build plate is black. The initial start of the new beta shows the grid but after closing the dialog its black again.

[image]

[image]

3Delights, 2022-08-17:
Rolling my AMD GPU drivers back to version 22.6.1 (June 2022) fixed the problem for now...

hejllukas, 2022-08-19:
We tried to investigate this issue, and it seems that the two latest AMD GPU drivers, 22.8.1 and 22.7.1, probably contain some bug that is causing missing bed texture. For some reason, those new drivers have an issue when mipmaps are generated manually on CPU sides combined with texture compression.

We generate mipmaps manually on the CPU side because we previously had issues on some GPUs with generating mipmaps on the GPU side.
And for some reason, those new drivers stopped accepting texture mipmaps generated on the CPU side.

So we made a workaround that for AMD GPUs with drivers newer than 22.6.1 (the latest working driver), we generate mipmaps on the GPU side, which seems to work correctly now.
This workaround will be part of the upcoming PrusaSlicer 2.5.0-rc1, so the texture of the bed should start showing again.

For older PrusaSlicer versions, the only way (at the moment) to get texture showing is to downgrade the AMD driver to 22.6.1 or 22.5.1 (WHQL).

bubnikv, 2022-08-19:
Please try with PrusaSlicer 2.5.0-rc1. The issue should be resolved.

mklengel, 2022-08-19:
In the end generating textures on the GPU side should be preferable. Not only because of the possible bug but also for better performance. Right?

bubnikv, 2022-08-19:
We cannot know how the driver works. If we ask the driver to create mipmap,
we don't know whether the mipmap will be constructed on CPU or GPU. It is
driver's responsibility.

mklengel, 2022-08-19:
I can confirm PrusaSlicer 2.5.0-rc1 shows the texture again.

Yornik, 2022-08-20:
Fixed my problem can comfirm thank you for your hard work.

kubispe1, 2022-09-17:
Ok guys, thanks for confirmation of bugfix. I am closing the issue.

closed 2022-09-17 by kubispe1

Bruellmuecke112, 2022-10-31:
I do have the problem again with prusaslicer 2.5.0 and the latest driver AMD-PRO-Edition 22 Q3 I'm not using the adrenalin edition but the pro edition.

I'm a little confused why the bugfix is not working here.

[image]

PrusaSlicer
Version:   2.5.0+win64
Build:     PrusaSlicer-2.5.0+win64-202209060714

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (Erzeugungsversion 22000), 64-bit Edition
Total RAM size [MB]: 33,689MB
OpenGL installation
GL version:   4.6.0 Compatibility Profile Context .220823
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon RX 6600 XT
GLSL version: 4.60
<details>
<summary>Installed extensions:</summary>
GL_AMD_blend_minmax_factor
GL_AMD_bus_addressable_memory
GL_AMD_depth_clamp_separate
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_pinned_memory
GL_AMD_sample_positions
GL_AMD_shader_trinary_minmax
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 180 lines cut ...]
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_NV_blend_square
GL_NV_copy_image
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_OES_EGL_image
GL_OES_draw_texture
GL_OVR_multiview
GL_OVR_multiview2
GL_OVR_multiview_multisampled_render_to_texture
GL_S3_s3tc
WGL_EXT_swap_control
</details>

Darthmonkey, 2022-11-01:
I too, have the 22.Q3 release for my Radeon RX 580. It's been working fine up to the update, when it went black.
Using the '--sw-renderer' work around does allow it work, but it's not a fix.

Arksine, 2022-11-06:
I have the same issue with an RX 570, although as mentioned this its pretty clearly a bug on AMD's side.  It seems that with every new driver release they break things in a new and interesting way.  Credit to Prusa for trying to work around it, however at some point AMD needs to fix this issue on their end.

infinitemachinery, 2022-11-22:
I'm having the same issue. WX 3200 series video card. Installed the latest drivers, problem still persists.

jandk, 2022-11-22:
Running latest Radeon PRO drivers, 2022Q4, still the same issue.

evolwun, 2022-11-26:
I am running a Radeon RX 5700 on driver ver 22.11.1 and am now experiencing the same issue on PS 2.5.0.

adelyser, 2022-12-03:
I am also having this issue, with a RX5700XT and the 22.11.1 drivers, using PS 2.5.0. It's certainly annoying.

TEAMMYKUNG, 2022-12-30:
> I am also having this issue, with a RX5700XT and the 22.11.1 drivers, using PS 2.5.0. It's certainly annoying.

running the program with  "--sw-renderer" parameter
It worked for me , RX5700xt 22.11.2 driver