prusa3d/PrusaSlicer #8511: arachne engine induces hight stuttering

MagicAlien, 2022-07-15:
### Description of the bug

When Arachne engine is on, the produced gcode induces the printer into what looks like high stuttering, when printing the outer perimeter of the inside (3d model attached). The resulting surface printed quality is terrible, and of course, the problem increases the printing time.
Note:
-- Observing the printer printing, you would see that while the print head is stuttering the extruder does many micro retraction and deretraction
-- My printer use TMC2208 stepper drivers for the axis motors and the A4988 for E
-- Linear Advance is active (K0.85) but no improvement deactivating it
-- Lowering the "slice resolution" and the "gcode resolution" until 0.037 won't influence the problem
-- I use some custom extrusion width, but changing them from custom to "0" won't influence the problem
[image]
[image]
[image]

### Project file & How to reproduce

[ring-test.zip](https://github.com/prusa3d/PrusaSlicer/files/9121518/ring-test.zip)
In the attached zip file you will find the 3D model and the 2 gcode (with and without arachne) used for the test, so you could analyze all the parameters too.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha2+win64

### Operating system

Windows 11

### Printer model

Anycubic Kossel Linear Plus with marlin 2.0

hejllukas, 2022-07-15:
Thank you for reporting this issue. Today we released [2.5.0 Alpha 3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha3), in which we fixed the issue #8455, which may be related to yours.

I also looked at the GCode you provided, and there are a lot of short extrusions in it. These short extrusions are not as short as in the #8455, but maybe they trigger firmware retractions or something.
In 2.5.0 Alpha 3 we fixed the #8455 issue I mentioned above. I tried to slice the attached 3MF in 2.5.0 Alpha 3, and the short extrusions are gone even GCode is much smaller. So I think that your issue could also be fixed in 2.5.0 Alpha 3.

Could you please try 2.5.0 Alpha 3 and report back whether your issue is still present or not?

MagicAlien, 2023-04-22:
Version: 2.6.0-alpha6+win64
Build: PrusaSlicer-2.6.0-alpha6+win64-202303311519

The problem is still here

[contenitor-dx.zip](https://github.com/prusa3d/PrusaSlicer/files/11301036/contenitor-dx.zip)

closed 2026-09-29 by prusaslicer-bot[bot]