prusa3d/PrusaSlicer #8512: Seams set to Rear become curvy instead of straight? Could we toggle to old seam behavior?

dougemon, 2022-07-15:
### Description of the bug

Hi there. I usually set my seam position to Rear when I print upright models. The seam position in 2.4.2 usually follows the line segments that are at the rear position.
In 2.5.0, it seems to smooth the seam position over the z axis, creating curves where it used to be straight.
Could we have a toggle to go back to the old seam behavior? Or possibly to be able to tweak the smoothness setting?
I am attaching 2 images and 2 3mf files showing the difference
[image]
[image]

### Project file & How to reproduce

[IceCream_RearSeam_2_4_2.zip](https://github.com/prusa3d/PrusaSlicer/files/9123103/IceCream_RearSeam_2_4_2.zip)
[IceCream_RearSeam_2_5_0alpha2.zip](https://github.com/prusa3d/PrusaSlicer/files/9123104/IceCream_RearSeam_2_5_0alpha2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha2+win64-202206281406

### Operating system

Windows 10

### Printer model

Original Prusa i3 MK3S+ with Bondtech Extruder Upgrade

gordo3di, 2022-07-19:
This was fixed in 2.5 alpha 3.

See https://github.com/prusa3d/PrusaSlicer/issues/8443 and https://github.com/prusa3d/PrusaSlicer/issues/8504

dougemon, 2022-07-26:
Hi there,
I installed 2.5.0 alpha 5 and it's still wiggly. Can there be an option to change back to the old behavior? Or possibly let the user change the smoothness factor
[image]

Snuff1eupagus, 2022-08-01:
I'm in support of this suggestion, I'm still seeing wiggly unwanted seams in the alpha 3 version. They do not look good, it would be better if we could decide or just leave them straight, as they were in 2.4.x

mmcglumphy, 2022-09-03:
I make lots of pots and containers, and the current seams are very unreliable when a perfectly straight line is desired. The new seam smoothing is wiggly for me in almost all cases. My current workaround is to manually and painstakingly paint a straight line. (A line drawing tool for seams would be stellar.) 

Here is one of my production pots Sliced in the latest RC2. In previous Slic3r versions this seam is completely straight.

[image]

Attached is a quick and dirty example project. I'm surprised that more people aren't reporting this. 

[image]
[shape.zip](https://github.com/prusa3d/PrusaSlicer/files/9483077/shape.zip)

kubispe1, 2022-09-07:
This issue is fixed in PrusaSlicer 2.5.0-rc2, so I am closing it. In case, let mi know.

kubispe1, 2022-09-07:
Ok, in 2.5.0-rc2 we upgrade seam behavior, but I see it is not still ideal. so after crosscheck I open it again.... Sry

Snuff1eupagus, 2023-02-03:
Fix the Seams please, just put them back the way they were in 2.4 and things will be fine. They are really bad in 2.5 and apparently going to be left this way in 2.6 if the alpha is anything to go by. 

I'm stuck using 2.4 because of this. I would love to know why they felt they needed to be messed with in the first place.

Jan-Soustruznik, 2024-09-17:
Hello, @dougemon , @Snuff1eupagus  I Closed this issue as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8443 

https://github.com/prusa3d/PrusaSlicer/issues/8443#issuecomment-2353205710
> Hello, @ramosglauco , Could you check the seam placement, with new PrusaSlicer release, 2.8.1-rc1 (link to download: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1-rc1).
> 
> Seam algorithm was enhancement now the placement of seams with specific object geometry should be improved. "Rear" placement was changed back before slicer 2.5.0

closed 2024-09-17 by Jan-Soustruznik