prusa3d/PrusaSlicer #8513: Flagged as ransonware in TrendMicro

terryrobison, 2022-07-16:
### Description of the bug

Flagged as ransonware in TrendMicro when I ran the program.  I stopped at this point

### Project file & How to reproduce

Run program normally

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha3+win64

### Operating system

Windows 10

### Printer model

E3d ToolChanger Duet 2 RRF 3.41

terryrobison, 2022-07-17:
I deleted the download, and downloaded again.  I did not get any warning from TrendMicro this time.  Time to experiment now.  Regards,

lukasmatena, 2022-07-17:
It is likely that it was a false alarm rather than actual danger. We already discussed antivirus warnings in the past, see e.g. https://github.com/prusa3d/PrusaSlicer/issues/6878#issuecomment-914456002.

Closing.

closed 2022-07-17 by lukasmatena