prusa3d/PrusaSlicer #8514: Access Violation slicing - BLOKHEAD.hf.3.30.21

leopinnheiro, 2022-07-16:
### Description of the bug

I get an access violation exception when slicing the model {}
I downloaded the file from **BLOKHEAD.hf.3.30.21.stl**
Imported to **[Prusa Slicer 2.5.0-alpha3+win64]**
When I try to slicing I get access violation.

[BLOKHEAD.hf.3.30.21.zip](https://github.com/prusa3d/PrusaSlicer/files/9124831/BLOKHEAD.hf.3.30.21.zip)

If I change the z position of the model to 6, I can slice
[image]

### Project file & How to reproduce

1. Open Attached BLOKHEAD.hf.3.30.21.3mf
2. Attempt to Slice
3. Observe Exception

[BLOKHEAD.hf.3.30.21.zip](https://github.com/prusa3d/PrusaSlicer/files/9124831/BLOKHEAD.hf.3.30.21.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha3

### Operating system

Windows 11

### Printer model

Ender 3 V2

kubispe1, 2022-07-16:
In my case, I reached same result for your setup (BLOKHEAD.hf.3.30.21.3mf). Set perimetr generator to Classic instead of Arachne (Print Settings/Layers and perimetres / Advanced / Perimetr generator). Then It works in my case.

VegetableJoe, 2022-07-16:
In my case, I reached same result for your setup (BLOKHEAD.hf.3.30.21.3mf). Set perimetr generator to Classic instead of Arachne (Print Settings/Layers and perimetres / Advanced / Perimetr generator). Then It works in my case.

leopinnheiro, 2022-07-17:
It works like magic, thanks a lot.

lukasmatena, 2022-07-17:
The solution described by @VegetableJoe just walks around the issue. We don't want Arachne to crash. Reopening.

hejllukas, 2022-07-21:
Thank you for reporting this issue. It helped us discover a weird issue of the Voronoi diagram generator library. With commit https://github.com/prusa3d/PrusaSlicer/commit/63c66f4f189b6f3d3a3b95ddd158c66ee5166420, we introduced the workaround for those types of issues. So in master_250, it is fixed, and this fix will be part of the next alpha or beta.

closed 2022-07-21 by hejllukas