prusa3d/PrusaSlicer #8515: Infill in ball object not added

Yourigh, 2022-07-16:
### Description of the bug

Hi, 

I found an issue with infill generation. At any setting (any infill or density) The ball model is not adding infill into ball part. Infill is added only to the top. (pic2).

Model can be obtained here: https://www.printables.com/model/237191-ballsocket-for-modular-mounting-system/comments

(pic1)
[image]
(pic1=2)
[image]

### Project file & How to reproduce

1. Add model to prusaslicer
2. Slice
3. Observe infill

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Windows 11

### Printer model

Prusa Mini

Alpacad, 2022-07-16:
Hi @Yourigh 
This is not an issue with PrusaSlicer, **the model is empty**, so PrusaSlicer will see the inside as external perimeters.
If you cut the model in the slicer you can clearly see that it was designed like this.

andy5macht, 2022-07-18:
Yes this is no bug. You should use Support instead of Infill here.

closed 2022-07-18 by stepikovo