prusa3d/PrusaSlicer #8516: compile the core to WebAssembly

smallstepman, 2022-07-16:
How feasable it would be to compile this to webassembly (leaving GUI aside - i.e. compiling only the core/backend)? Additional GUI in JS would have to get written.

This would setup a path which leads to having PrusaSlicer embedded inside other web clients, like octoprint, fluidd, mainsail etc.

Tristaan, 2022-12-15:
Hello @smallstepman!
I too am interested in this "feature-request". But as PrusaSlicer has many dependencies, which also need to be compiled to WebAssembly it is a hard task to do. Maybe someone could shed more light on the situation.

schlaegerz, 2025-01-27:
Just curious, did you ever figure this out?

Tristaan, 2025-01-29:
> Just curious, did you ever figure this out?

No, iirc GUI lib issues, intertwined a bit. Also file management.

schlaegerz, 2025-02-20:
I have been able to get it compiled to wasm
 
https://github.com/ToyboxZach/PrusaSlicer

I had to do a decent amount of hacking and its not quite a reproducible build, but I tried to leave me thoughts there in case someone with more CMake experience wants to tackle it.

Big keys is I did not compile the GUI and removed the SLA support since those had a large number of dependencies.

But this compiles and I have been able to run it in the browser

closed 2026-09-29 by prusaslicer-bot[bot]