prusa3d/PrusaSlicer #8519: Added support for building with MSBuild Tools

StephS, 2022-07-17:
I have worked to modify the build_win.bat to not only support Visual Studio 2019, but also MSBuild Tools 2019. MSBuild Tools has all of the same compiler tools as Visual Studio, but without requiring a Visual Studio install. This can help allow a leaner dev environment for those who do not wish to install Visual Studio.
The changes include modifying the lines where vswhere.exe is used, and some switches for testing the functionality (I would have added command line flags but I don't understand the parsing section).

You can view the changes on my fork here: https://github.com/StephS/PrusaSlicer/tree/MSBuild
To Test, install MSBuild Tools 2019 (I think I also needed to install powershell). Then edit the line "SET PS_USE_MSBUILD=" to "SET PS_USE_MSBUILD=1".
I have been able to test on my own system with both Visual Studio and MSBuild installed.

With approval I can create a pull request.

lukasmatena, 2022-07-19:
Thanks. The build script is @jschuh 's baby. He would be the most competent to review the changes.
@jschuh Would you mind looking into this, if / when you have the time and the right mood?

jschuh, 2022-07-21:
Seems reasonable to support MSBuild. My original goal was just to make it as easy to spin up a custom build on Windows as it is on Linux, but I don't think anyone would object to supporting a broader range of tooling.

You can take a look at commit cfdf7d2a00c3979ae52946e2ae0a920b8e8d514e for the simplest possible example of adding a command line parameter. That should work as a template for that part of your proposed change.

I do have a few more suggestions on the implementation:

- Might as well just always fallback to checking for MSBuild and using it if available, since that's similar to the behavior with multiple installed MSVC builds/versions; then the switch would control whether the caller is explicitly setting it to MSBUILD or MSVS.
- It looks like this changes the version handling in an incompatible way (line 98 in your fork); right now v16 is the max supported version, because the deps won't build with v17, and we don't want to break that behavior.
- I wouldn't include your logging changes in the PR for MSBuild support; Windows bat script is hard enough to follow without trying to do too many things in one change (although I wouldn't complain about a standalone PR that adds selectable logging levels).

StephS, 2022-07-24:
Thanks, I created a pull request https://github.com/prusa3d/PrusaSlicer/pull/8554
Regarding the version handling, v17 is excluded (it is set to search for <17). This is the same behavior that was seen here: https://github.com/prusa3d/PrusaSlicer/pull/8554/files#diff-bf6b22b4e8a901118a4bd8786d11c2e06a0553e374dfd2c4dcdbbce795003a8bL80

StephS, 2022-08-04:
Pinging @jschuh
I think it's ready, it's difficult to test with msbuild only and MSVS only installs, but the command line switch works successfully to switch between the two.

closed 2026-09-29 by prusaslicer-bot[bot]