prusa3d/PrusaSlicer #8524: Arachne - broken/chunked inner perimeters

Kuba2pl, 2022-07-19:
### Description of the bug

When printing my own models that have the "ideal" thickness (as specified in "Vertical Shells" note - 1.67 mm - screenshot 1), the inner perimeters are sometimes printed with chunks that are thinner/wider than the rest of the perimeter (screenshots 2 & 3). 

Those chunked sections are also affecting the quality with higher print speeds, as the nozzle often slows down in those regions.

The classic perimeter generator creates continuous perimeters without any problems mentioned above.
[image]
[image]
[image]

### Project file & How to reproduce

[Examples.zip](https://github.com/prusa3d/PrusaSlicer/files/9140540/Examples.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha3+win64

### Operating system

Windows 10

### Printer model

MK3S+ / MINI+

Kuba2pl, 2022-07-22:
[3mf files example.zip](https://github.com/prusa3d/PrusaSlicer/files/9168136/3mf.files.example.zip)
Starting G-code was modified so it'd be easier to see the problem.

kubispe1, 2022-09-15:
I also can not reproduce it. Can you try it again with new 2.5.0 ?

Misfi, 2022-09-19:
I had a very similar issue, and I can still reproduce it with OP's example.

@kubispe1 it's not visible on every layer, but it is there for sure. Make sure you remove the purge line to make the extrusion width difference more visible - otherwise it's just one shade of dark blue...

[image]

DragonFinlay, 2022-09-22:
Seems like the cause of this issue is the object's geometry/topology. These slightly thinner/thicker lines only happen at specific heights, which got me thinking.

I gave the cylinder a quick retopology in Maya and the issue is not happening on the new cylinder anymore.

(left -> original, right -> retopologized)
[image]
[image]

So whatever modeling software is being used, try tinkering with the export resolution of the model. If you need a consistent extrusion width throughout the entire print you might need a higher resolution.

Closing this issue as it's not connected to PrusaSlicer.

closed 2022-09-22 by DragonFinlay