prusa3d/PrusaSlicer #8527: Inconsistent naming of files for release on GitHub

darrelldudics, 2022-07-20:
### Description of the bug

Can we have consistent naming of files? I know it's such a small detail, but I had to do a double-take because there is no file containing "osx" in it.

It's currently as follows:

PrusaSlicer-`<version>`+`<os_name>`-`<architecture>`-`<date_time>`.`<ext>`

Except for OSX, the <os_name> is left out and is only named as:

PrusaSlicer-`<version>`+`<architecture>`-`<date_time>`.`<ext>`

[image]

### Project file & How to reproduce

As seen on https://github.com/prusa3d/PrusaSlicer/releases

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5 Alpha 3

### Operating system

OSX

### Printer model

Prusa i3 MK3S

thisiskeithb, 2022-07-23:
How about switching to “macOS” since it was introduced in 2016 🙂

darrelldudics, 2022-07-30:
> How about switching to “macOS” since it was introduced in 2016 🙂

Not sure what you mean by that since "MacOS" is the same thing as "OSX".

thisiskeithb, 2022-07-30:
It was renamed in 2016, so it would make sense to use the new name here.

lukasmatena, 2024-02-19:
Solved, closing.

closed 2024-02-19 by lukasmatena