prusa3d/PrusaSlicer #8529: 【MMU Wipe options】Wipe into object should work with no sparse layer as well

5jvm0u4, 2022-07-20:
**Is your feature request related to a problem? Please describe.**

Wipe into object doesn't work like wipe tower that can do no sparse layer wiping. If it does, it will save a lot of filament.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
[image]

lukasmatena, 2022-07-21:
Please, post a 3MF file.

5jvm0u4, 2022-07-21:
[Uploading No sparse layer.zip…]()

stepikovo, 2022-07-21:
@5jvm0u4 Unfortunately that's a bad link

lukasmatena, 2022-07-21:
You only posted a link to this issue, not a 3MF file. I assume you sent the post before the file uploaded.

5jvm0u4, 2022-07-21:
I'm very sorry. Here's the file.
[No sparse layer.zip](https://github.com/prusa3d/PrusaSlicer/files/9157910/No.sparse.layer.zip)

lukasmatena, 2022-07-21:
Oh, I understand what you mean now. You are suggesting that even the layers of the sacrificial objects may be printed later.

We are currently not planning to implement this. It adds too much complexity for not that much value. For example, that object might need supports. When will they print? What if they are configured to have interface made of specific material? Who will switch to it and when? What about cooling slowdowns? Etc. It also increases possibility of collisions - in fact, more material might globally be thrown out after misconfigured failed prints compared to material saved by the others. We implemented it for the wipe tower because it is architecturally a bit aside from the rest of the G-Code generator, its shape is predictable, etc.

closed 2026-09-29 by prusaslicer-bot[bot]