prusa3d/PrusaSlicer #8530: 2.5.0 alpha 3 - Extremely slow inner perimeters and infill on some layers with ironing on

KimmoHop, 2022-07-20:
### Description of the bug

When ironing is off, objects are printed normally.
When ironing is on, layers 8 and 9 have very slow printing moves (0.4mm/s?). Noticed only when printer looked like dead, but it was moving in the end.

[image]

### Project file & How to reproduce

[Pill_Box_ironing_error.zip](https://github.com/prusa3d/PrusaSlicer/files/9149718/Pill_Box_ironing_error.zip)

Slice attached file. Layers 8 and 9 should show extremely slow movement

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha3+win64

### Operating system

Windows 10 Home 2H21

### Printer model

Voxelab Aquila, profile based on Ender 3

kubispe1, 2022-07-20:
There could be problem in unsuitable combination of ironing and value of pressure equlizer. In layer 8, there is a small surface where ironning is puts. Irroning has basicly very low volumetric pressure (~speed) so there is long length for transition to higher pressure (~speed) due to preassure equilizer. It also impacts layer nb.9... If it is acceptable for you, I recommed to increase Max volumetric slope positive and Max volumetric slope negative and try to tune it, or even switch off it (put there 0).

KimmoHop, 2022-07-21:
This is with 10mm3/s2 volumetric slope:
[image]

I usually don't use ironing, and in this case with Volcano hotend running hot (to get some layer adhesion with silk pla) the result was a bit coloured and some rough at the edges of the raised ridges, but otherwise quite nice ;)
But for those using ironing and wanting to use pressure equalizer, you really can't make them to select one or another, can you? I mean it's not grave important right now, but it's really bad to have such lines (0.5mm/s! You have to try it ;) ) with common-ish combination of settings.

With classic engine it's similar, but in some parts worse, e.g. layer 20 (19 and above). What does it slow down for?
Arachne: 
[image]

Classic: 
[image]

hejllukas, 2022-07-28:
Thank you for reporting this issue. It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/394494b3acb9bd888b88b4257f977c4844ea76e9 in master_250, and it will be part of the next alpha.

KimmoHop, 2022-07-28:
I think there's still something wrong on layers 8 and 9. e.g. move to second part after ironing -> starts at ~40mm/min

The second part, layers 19 and above, was accelerating bridge infill after solid infill. Blues are difficult to tell apart ;) - still why so different between engines?

hejllukas, 2022-08-09:
Sorry, I overlook it.
The reason why Classic and Arachne had different behavior was that Classic is producing a gap-fill, and there was a significant slow down before a gap-fill (as was with ironing), which was unintentional. It should be fixed with 7dac30714b4ebb29fd807fdeeb9db1e21452efbb, so the pressure equalizer should have similar behavior for Classic and Arachne now.
Also, the commit fixes an issue that there wasn't a deceleration before the bridge infill.

closed 2022-08-09 by hejllukas

KimmoHop, 2022-08-14:
Thanks! Seems to work correctly ;)