prusa3d/PrusaSlicer #8531: Total cost of print including power

zbuilder801, 2022-07-20:
It would be nice to have 2 more fields... one for the max wattage of the printer and another for the cost per kilowatt and these values used to determine a closer total cost estimate per print. Even though the total wattage of the printer is not always used, the extra cost calculated would make up for the time the printer is just turned on and not printing.

gudvinr, 2022-07-21:
How do you suggest to calculate that exactly so it won't be useless?

You need to consider different power profiles for heating bed and hotend which will be also different for different types of beds and different hotends.  
There are also stepper which draw power which not only depends on printer moves but also on set current which isn't always configurable from software, even.  
You also need to somehow figure out power profile for mainboard (and maybe SBC if you use e.g. raspberry pi).  
There is also power draw (albeit less significant) for peripherials and other stuff.

If you need to know how much power you use for a print your best option is some smart plug with power meter hooked to Home Assistant or something similar.

murk-sy, 2022-07-22:
I think having electrical running costs in the slicer is a bit overengineered - though depending on your printer, you could probably reach 15% of the material costs (with that being a pretty extreme calculation), not counting actively heated chamber printers.

However, having "Running cost per hour" and "Setup cost per started print" settable would make more sense to me, as you can then basically extrapolate those from all the other costs. With those 2 options you can then do a fairly comprehensive price (or cost) calculation. Ideally you'd have a separate filament cost multiplier too, but theoretically you can also just multiply it in the slicer - not exactly optimal though.

See also #3199

Hologos, 2022-08-08:
@zbuilder801 I am using smart plug (as @gudvinr suggested) and it work great. You have exact power consumption with all the peripherals (if you plug all of it to the plug). The only downside is you have to manually get the data.

guestisp, 2023-02-17:
SuperSlicer has this.
I don't think having a very accurate value is useful, you should know, more or less, the cost per kilowatt/h applied by your provider, then you add a small % to that hourly cost and you can obtain a rough indicator of print cost for customers.

In example: you know that the bad is 750w. The hotend is 50w. Running the printer for 1 hour is about 800w (probably less, because both doesn't run at full power every time). If your kwh cost is 0.10 euro, a 1h print would cost (800/1000)*0.10 = 0.08 euros (plus the material)

DaLiV, 2023-07-16:
So. i can suggest how to calculate.
All Params seems are mainly Printer-Config-Related
that can be done by prior measurings with Outlet Energy Meter (such you can frrely find on amayon/ebay and so on )  of used wattage in diffirent modes
1) Powered On - Idle = Approx. 9.5-10W (that must be substracted from each next step )
2) TurnOn Heating the Bed Only
  + measured heating time to 30->60C => takes 6Min:15Sec =375Sec and Outlet Energy Meter shows approx 250-220W /this dropping down in time/ (minus value from [1] => 240-210  but for calculation suggest tak 240Wh )  
=== **[ 240 * 375Sec/3600 Wh ]**
  + Temp supporting mode lokks like 15 Sec Off + 15 Sec On so K=0.5 and power at 220Wh 
=== **[ 220 * 0.5 * [(PrintTimeSec + ExtruderPreheatTime)/3600] ] Wh**
3) TurnOn Heating the Extruder Only (bed is off) - That Gives next values
  * PreHeating 30-190C = 2Min =  50-55Wh (-10 from [1] = 40-45Wh ) 
=== ExtruderPreheatTime=120Sec | **45 * 120/3600 Wh**
  * Supporting 190C = 28-33 (again -10 = 18-23Wh ) 
=== **23 * [PrintTimeSec/3600]**
4) Movements Power - approx 
**=== 50W * PrintTimeSec**

DaLiV, 2023-07-16:
So. i can suggest how to calculate.
All Params seems are mainly Printer-Config-Related
that can be done by prior measurings with Outlet Energy Meter (such you can frrely find on amayon/ebay and so on )  of used wattage in diffirent modes
1) Powered On - Idle = Approx. 9.5-10W (that must be substracted from each next step )
2) TurnOn Heating the Bed Only
  + measured heating time to 30->60C => takes 6Min:15Sec =375Sec and Outlet Energy Meter shows approx 250-220W /this dropping down in time/ (minus value from [1] => 240-210  but for calculation suggest tak 240Wh )  
=== **[ 240 * 375Sec/3600 Wh ]**
  + Temp supporting mode lokks like 15 Sec Off + 15 Sec On so K=0.5 and power at 220Wh 
=== **[ 220 * 0.5 * [(PrintTimeSec + ExtruderPreheatTime)/3600] ] Wh**
3) TurnOn Heating the Extruder Only (bed is off) - That Gives next values
  * PreHeating 30-190C = 2Min =  50-55Wh (-10 from [1] = 40-45Wh ) 
=== ExtruderPreheatTime=120Sec
    **45 * 120/3600 Wh**
  * Supporting 190C = 28-33 (again -10 = 18-23Wh ) - fluctuated, not off-on
=== **[23 * [PrintTimeSec/3600] Wh]**
4) Movements Power - approx ... also here can be put sum oif all other active power usage that can be used in cost-calculation
=== **50 * PrintTimeSec/3600 Wh**

guestisp, 2023-07-16:
way too complicated and useless, you don't need 0.1 cent precision

Just use heather+extruder watt for the full print time.

you have a 750w bed and 50w hotend and a 6 hours print? 6*800w

Hologos, 2023-07-16:
> way too complicated and useless, you don't need 0.1 cent precision
> 
> 
> 
> Just use heather+extruder watt for the full print time.
> 
> 
> 
> you have a 750w bed and 50w hotend and a 6 hours print? 6*800w

Sorry but your suggestion us wrong. The bed and heater don't go on full power all the time, that's what PID is for. It's turning it off and on all the time. For example - P1P printing PLA with set temperature 240C hotend and 65C bed eats about 130 Wh of energy (aka 0.13 kWh for most of the printing) with it's LED on.

guestisp, 2023-07-16:
i know that is wrong but if you have to sell prints, the extra amount is your earning thats why i consider 800w plain. to me, it cost less

guestisp, 2023-07-16:
more accurate calcs are not for the slicer...

DaLiV, 2023-07-16:
PID -> therefore exist 2 timings for each ... 
and mine calculation are "electricity consumption in Wh" ... which you has no calculated or measured ... 
surely can measure some months by printing 1000 items and you knew average ... 
but in my solution you can get this values in less than 30 minutes ...

murk-sy, 2023-07-16:
Energy consumption depends on actual target temperatures and the ambient temperature as well (even if enclosed but to a lesser degree). Very soon you get to the point of splitting hairs.

In my mind, overestimating energy usage is "good enough", as long as you pick a good baseline.

To demonstrate:
Power costs around 0,50€/kWh in the generally more expensive countries.
  a) MK3 power supply is 240W.
  b) MK3 heatbed + nozzle is 190W.
  c) Prusa [FAQ](https://help.prusa3d.com/article/faq-frequently-asked-questions_1932) lists power usage as "80W PLA and 120W ABS" - slightly old data but good enough.

Assuming a 10 hour print at that provided wattage, this means:
  a) 1,20 €
  b) 0,95 €
  c) 0,60 € ABS 0,40 € PLA

In that time, you've used (250% benchy 0,2 quality profile on mini, 9h31m) ~110g of filament, which, if you're using a fairly cheap one, at 15€/kg costs you 1,65 €.

So power cost is **not** insignificant. But in my opinion just using the heatbed+nozzle wattage is "good enough".
You're looking at your estimate being either 50% more or double the actual power usage during the print. If you're looking at the costs either for your personal use or your business, overestimating them is better. The downside of using this approach (from Prusa's standpoint) is that the printers then appear more energy wasteful than they are.

What it's not included in that estimate however is the idle energy consumption and also your slicing energy consumption. Is including that splitting hairs? Probably.

As a side note, when I account for energy costs in prints, I use ```print time * 2 * power cost * printer psu wattage```.
Is it accurate? No. But I'm not interested in calculating energy use. I'm interested in covering my energy costs and not having to adjust pricing for a while.

Hologos, 2023-07-16:
> PID -> therefore exist 2 timings for each ...
>
> and mine calculation are "electricity consumption in Wh" ... which you has no calculated or measured ...
>
> surely can measure some months by printing [...]

I wasn't reacting to you, can't you tell by the quotation?

Your calculation is as correct as is someone's recoring of all times necessary for your calculation. The deviation from your appoach against mine is negligible and is somewhat covered in that 10% for failed prints (as long as fail print exceeds 10%).

I made hundreds of prints measured with the meter and average was +-5%. That is enough accuracy, the difference is in few cents.

DaLiV, 2023-07-16:
sure - i already can estimate even without power calculated - that is as simple as set price for Material +11Eur/Kg (for 0.50Eur/Kwh)  and result estimation will be in the same 95% range

zbuilder801, 2023-07-17:
Yes, it is “way too complicated” but if Prusa does the calculation, then let it use 0.1 cent precision. After thousands of hours it will probably matter to someone who has a printing business and many printers working 24 hours a day, 7 days a week… especially in expensive power countries.

My question… is there any circuitry in the design right now that can be read by the Prusa OS to sense the power being used or will that required a new motherboard design?

closed 2026-09-29 by prusaslicer-bot[bot]