prusa3d/PrusaSlicer #8532: Build Issue: openvdb::v7_2::tools::FastSweepingDomain’ has not been declared

pmjdebruijn, 2022-07-20:
### Description of the bug

When building PrusaSlicer 2.5.0~alpha3 against openvdb-7.2.3 (Ubuntu 22.04 ships openvdb-8.1.0 by default, which results in issues due to being linked against a newer version of TBB, related to #7472):

In file included from /usr/include/openvdb/tools/Morphology.h:30,
                 from /usr/include/openvdb/tools/FastSweeping.h:47,
                 from ~/Packages/prusaslicer/PrusaSlicer-version_2.5.0-alpha3/src/libslic3r/OpenVDBUtils.cpp:17:
/usr/include/tbb/tbb_thread.h:21:145: note: ‘#pragma message: TBB Warning: tbb/tbb_thread.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.’
   21 | #pragma message("TBB Warning: tbb/tbb_thread.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                                 ^
~/Packages/prusaslicer/PrusaSlicer-version_2.5.0-alpha3/src/libslic3r/OpenVDBUtils.cpp: In function ‘openvdb::v7_2::Grid<openvdb::v7_2::tree::Tree<openvdb::v7_2::tree::RootNode<openvdb::v7_2::tree::InternalNode<openvdb::v7_2::tree::InternalNode<openvdb::v7_2::tree::LeafNode<float, 3>, 4>, 5> > > >::Ptr Slic3r::mesh_to_grid(const indexed_triangle_set&, const openvdb::v7_2::math::Transform&, float, float, float)’:
~/Packages/prusaslicer/PrusaSlicer-version_2.5.0-alpha3/src/libslic3r/OpenVDBUtils.cpp:87:25: error: ‘openvdb::v7_2::tools::FastSweepingDomain’ has not been declared
   87 |         openvdb::tools::FastSweepingDomain::SWEEP_LESS_THAN_ISOVALUE);
      |                         ^~~~~~~~~~~~~~~~~~
~/Packages/prusaslicer/PrusaSlicer-version_2.5.0-alpha3/src/libslic3r/OpenVDBUtils.cpp:92:25: error: ‘openvdb::v7_2::tools::FastSweepingDomain’ has not been declared
   92 |         openvdb::tools::FastSweepingDomain::SWEEP_GREATER_THAN_ISOVALUE);
      |                         ^~~~~~~~~~~~~~~~~~

This issue was introduced in:
https://github.com/prusa3d/PrusaSlicer/commit/6d349dd3a794caed16a2bc4046025d34d33fdd90

I'm guessing this is relevant:
https://github.com/AcademySoftwareFoundation/openvdb/commit/c4e2ca82c404a6723c98c5e186861599747ff6b7
Which means PrusaSlicer needs at least OpenVDB 8.2.0 because of the FastSweepingDirection -> FastSweepingDomain name change.

I hope you'll consider ifdeffing that, so PrusaSlicer remains fairly easy to build across a wide variety of platforms.

### Project file & How to reproduce

Build against any OpenVDB version earlier than 8.2.0

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0~alpha3

### Operating system

Ubuntu 22.04

### Printer model

N/A

peter-kutak, 2022-09-29:
same on Debian 11
maybe at least adjust CMakeLists.txt to reflect required version

tamasmeszaros, 2023-12-18:
As of 374cb237ed6c8c6e093ac9f2f03c171bf527e7be, OpenVDB is updated in our dependencies to 8.2 and the build issues were also resolved. I'm closing this issue as the problem is no longer present in our codebase.

closed 2023-12-18 by tamasmeszaros