prusa3d/PrusaSlicer #8534: Arachne causes stuttering extruder on curves, rough surface with klipper.

napter, 2022-07-20:
### Description of the bug

When printing with klipper firmware the surface of the print was rough, exactly how it looks when the gcode is too large and printing stutters. Similarly, the issue only happened on curves. However, the same file was printing ok on other printers. 

Reducing the STL nodes didn't help. Increasing the gcode resolution didn't help. Changing speed didn't help. The artifacts would repeat in the same spots. When watching the printer it would vibrate when printing on the curves (I believe it was the the extruder stuttering but it was difficult to tell if it was the extruder or a motion axis). After determining it was not a resolution issue, finally, I thought perhaps it is Arachne. I printed the exact same thing, switching perimeter generator to classic and the issue disappeared.

Arachne print:
[image]

Classic print:
[image]

Side by side:
[image]

Notice the bumps on the curves. They print consistently on every print with arachne but not with classic. I've only experienced this issue with klipper. 

### Project file & How to reproduce

[arachne issue.zip](https://github.com/prusa3d/PrusaSlicer/files/9152609/arachne.issue.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5 alpha 2

### Operating system

Windows 10

### Printer model

Ender 5 Plus running Klipper

bubnikv, 2022-07-21:
Would you please test with alpha3?

From the changelog:
Release PrusaSlicer 2.5.0-alpha3 · prusa3d/PrusaSlicer (github.com)
<https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha3>

   - When using the Classic perimeter generator, consecutive segments of
   gap fill are now merged when they have the same direction and don't differ
   in width (or only differ slightly). This reduces size of the G-Code and
   makes printer stuttering less likely.

dartrax, 2022-07-22:
> When using the Classic perimeter generator

But @napter has this problem with Arachne, not with classic…

hejllukas, 2022-07-22:
@dartrax 
It was just a wrong quote from the changelog (actually, gap fill and Arachne share some parts of the code, so fixing one will fix a similar problem for the other).
Instead of that, it should be a citation of this that is definitely related to this issue:
> Arachne perimeter generator sometimes produced extremely short segments even when there was no need. These extrusions could have been incorrectly shown as deretractions and also could have caused extremely short and counter-productive wipe moves #8455

@napter
I tried to compare G-code for your attached 3MF from 2.5.0-alpha2 and 2.5.0-alpha3. G-code from 2.5.0-alpha2 contains a lot more short extrusions than G-code from 2.5.0-alpha3. Even a comparison of G-code sizes shows that 2.5.0-alpha3 produce G-code that is about 25% smaller than G-code from 2.5.0-alpha2.
So I think those short extrusions caused your issue, but the issue with short extrusions was resolved in 2.5.0-alpha3. A similar situation was also in issue #8511, and that issue was resolved with 2.5.0-alpha3.

So I think that this issue is also resolved in 2.5.0-alpha3. So please try [2.5.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha3) and report back whether your issue is still present or not.

napter, 2022-07-22:
Ok will do

kubispe1, 2023-02-20:
Please @napter can you retest the issue. In case please check with new  PS 2.6.0-alpha4. Hope this issue has been resolved to your pleasue :D ...
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha4

kubispe1, 2023-02-20:
Please @napter can you retest the issue? In case please check with new  PS 2.6.0-alpha4. Hope this issue has been resolved to your pleasue :D ...
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha4

lukasmatena, 2023-07-10:
No response, should be fixed. Closing.

closed 2023-07-10 by lukasmatena