prusa3d/PrusaSlicer #8536: Support material blocker does not behave when trying to print 'piano hinge' type feature - repeating unsupported vertical holes

n8maher, 2022-07-20:
### Description of the bug

I would like to print a 'piano hinge' type feature without the need to manually clean supports from inside the axle. The part has a continuous thru hole running the entire length of the part:
[image]
When sliced with Supports: Everywhere, every cylinder except the topmost one is filled in:
[image]
[image]
I would expect that adding a support blocker like this would fix this:
[image]
But the slicing results are unchanged. The blocker works fine for a single cylinder, but breaks on the repeated cylinders. I wonder if the support material interface requires the column of support material and overrides the support blocker.
[image]

This isn't the real part, but it breaks the same way.

Thanks for all your hard work, development team!

### Project file & How to reproduce

[PianoHingeSupportBlockerConundrum.zip](https://github.com/prusa3d/PrusaSlicer/files/9154047/PianoHingeSupportBlockerConundrum.zip)
Open file and slice it. Support material will be present in the cylinders even though a support material blocker is present.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+win64

### Operating system

Windows 10

### Printer model

Original Prusa MK3S, MMU filament sensor, no MMU :-)  (Yes, loading material is weird)

lukasmatena, 2022-07-21:
@n8maher This is not how support blockers work. They are used not to mark places, where you want not support extrusions, but to mark areas on the model (!!!) which you do not want supported. In your case, there is no object inside the cylinder, so supports are generated as they would be without it. The algorithm finds places that need supports and builds them - and because it sees the holes, it uses that space.

If the hole was small, you could increase `Print Settings->Support Material->XY separation between...`, which would make the support generator reject the hole as too small to be used. In your case you would have to increase it a lot, which would possibly create issues elsewhere. You can try it though.

stepikovo, 2022-07-21:
Setting the Pattern spacing to 0.5 mm and increasing the diameter of the blocker to 3 mm, you can achieve the desired outcome.

closed 2022-07-21 by stepikovo

n8maher, 2022-07-21:
Ahh that makes sense. Thank you @lukasmatena and @stepikovo for your help!