prusa3d/PrusaSlicer #8537: 3mf file doesn't include unsave changes in preset

5jvm0u4, 2022-07-21:
### Description of the bug

While saving project in 3mf format, I was expecting it to store my modification in the setting so I don't need to create a new preset every time I make a new project.
Like support setting, speed setting, purge volume setting, etc.
The first screenshot is the original project I was working on, and the second screenshot is what I saw when loading the project's saved file.
[image]
[image]
Even the object's position hasn't been saved.
Sometimes when opening the 3mf file it will show up an error dialog saying something like "one line can't be analyzed and will be ignored" but I failed to capture that.

### Project file & How to reproduce

[Statue.zip](https://github.com/prusa3d/PrusaSlicer/files/9157210/Statue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2 as well as 2.5.0 alpha 2

### Operating system

windows 11 with system language set to triditional Chinese

### Printer model

not relevent

5jvm0u4, 2022-07-21:
Update：
I found that while using add model button, this issue happens
[image]
but when open project is used instead, everything works as intended.
[image]

5jvm0u4, 2022-07-21:
[image]
I got the error message, I think it's because of the traditional Chinese file name
[漩渦漏斗V5.zip](https://github.com/prusa3d/PrusaSlicer/files/9158151/V5.zip)

stepikovo, 2022-07-21:
@5jvm0u4 Open Project (Ctrl + O) and Add/Import (Ctrl + I) are different things.

I was not able to reproduce this error on Win, we will check it on MacOS
> [image] I got the error message, I think it's because of the traditional Chinese file name [漩渦漏斗V5.zip](https://github.com/prusa3d/PrusaSlicer/files/9158151/V5.zip)

closed 2026-09-29 by prusaslicer-bot[bot]