prusa3d/PrusaSlicer #8538: Y and Z for modifier settings are swapped

patmis67, 2022-07-21:
### Description of the bug

In the modifier settings window, the position fields for Y and Z are swapped.

[image]

### Project file & How to reproduce

Create a modifier and try to position it in the settings. Only the fields for Y and Z are reversed.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2 / Win 64

### Operating system

Windows 11

### Printer model

Prusa Mini

stepikovo, 2022-07-21:
Please provide the project file saved as a 3MF. You probably have the model rotated

patmis67, 2022-07-21:
here is the 3mf:
[3M_Serie_891_50P_v2.zip](https://github.com/prusa3d/PrusaSlicer/files/9158433/3M_Serie_891_50P_v2.zip)

That is correct. I have rotated the model and if I change the coordinates to **local coordinates**  it fits. Sorry for the mistake and thank you for your support.

closed 2022-07-21 by patmis67