prusa3d/PrusaSlicer #8540: unnecessary travel between perimeters

andy5macht, 2022-07-21:
### Description of the bug

travel moves are generated where no movements are necessary within the permeter

### Project file & How to reproduce

[image]
[Stopfen_45_50_10.zip](https://github.com/prusa3d/PrusaSlicer/files/9158517/Stopfen_45_50_10.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha3

### Operating system

Windows 10

### Printer model

Own MKC2

stepikovo, 2022-07-25:
It's due to the fact the seam was set as _Random_, which makes even the internal seam at random locations, improving water-tighteness. When you set it as _Aligned_ the issue is gone.
Can we close this issue?

stepikovo, 2022-07-26:
Unfortunately making the internal seam aligned to the random external seam is not that simple in the current version, but we will consider it.

DrLex0, 2022-09-03:
I would greatly appreciate it if before the final 2.5.0 release, the ‘Random’ setting could be restored to how it behaved before, and this new setting would be renamed to “watertight” or something, if that really is the intention of randomizing the starting point of each contour even within the same layer. But, perhaps you should not rely on randomness to obtain watertight seams. The staggered seams that were introduced in 2.4.0-beta1 were a much better solution to improve watertightness, because even with random placement there is still a chance that starting points will coincide. Unfortunately the staggered placement was dropped again in 2.4.1-beta1 due to “unnecessary jerking,” but I deem all these pointless travel moves to be a much worse issue.

Webovani, 2022-11-05:
I am using random seam alignment while printing PVB to be smoothed after. That way the seam artifacts disappear completely - or at least seem "organic" enough that I cannot identify them. It doesn't work that well with aligned seams - there is still a visible bump line where the seams were.

But now this adds too much unnecessary travel time and causes significantly more stringing. I cannot afford to use it and I will have to rollback to previous version.

animatorgeek, 2025-02-25:
I'm using 2.9.0 and it appears that this issue has been fixed. Can we mark this resolved?

That said, I actually WANT the behavior described here. I'm constantly striving for watertightness and I've never gotten it without overextrusion.

andy5macht, 2025-02-27:
Yes its true. The Issue is gone with the newer version.
Thank you!

closed 2025-02-27 by andy5macht