prusa3d/PrusaSlicer #8542: MMU - Allow user to choose initial extruder but then also change extruder at layer height

DeanDavis, 2022-07-21:
When using PrusaSlicer (2.4.2) in full MMU mode (not single) you can't seem to set the starting extruder when doing a color print where the color changes come via color change on layer height change.

To describe in another way if, in a MMU print you specify a specific extruder number in the right side panel for the print you cannot specify a color change to a different extruder using the height slider.

With pictures...
[image]
[image]

**Describe the solution you'd like**
Allow user to choose a specific extruder for the beginning of the print but then allow user to change extruder at specific layer heights.

**Describe how it would work**
MMU would initialize on choosen extruder at beginning of print then change extruder to the one choosen by user using the layer height slider

**Describe alternatives you've considered**
Right now I just change the GCODE for the initial extruder choice.

stepikovo, 2022-08-04:
Can you please share with us the printing project as 3mF?

DeanDavis, 2022-08-04:
[Extruder Change Test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9260815/Extruder.Change.Test.3mf.zip)

Looks like it Height Range Modifier related.
In the attached example if I have a HRM it will not allow me to change extruders at a height but if I remove the HRM it will let me change extruders.

stepikovo, 2022-08-18:
Thank you for the project file, looks like it is caused by inconsistency in the extruders, you must have the same extruder even in the modifier, which, unfortunately, you still have it set to _default_ (1).

DeanDavis, 2022-08-18:
How do I set it to anything else? If I click in word "Default" next to the Height Range Modifier, left or right click, I don't get any menu allowing me to change it.

stepikovo, 2022-08-22:
First, you select the object and the second click then opens the menu.
Let me know how it works so we can close this issue.

YuSanka, 2022-08-22:
@DeanDavis is right. There is OSX specific issue. It's fixed and probably will be a part of a final release.

DeanDavis, 2022-08-23:
> @DeanDavis is right. There is OSX specific issue. It's fixed and probably will be a part of a final release.

Just as a matter of FYI, I just downloaded and tested 2.5.0-rc1 and this issue is not fixed in that release.

YuSanka, 2022-08-23:
> 
> Just as a matter of FYI, I just downloaded and tested 2.5.0-rc1 and this issue is not fixed in that release.

This fix is not in master_250 jet ;)

DeanDavis, 2022-09-12:
Just a FYI comment for anyone following. This was not fixed in 2.5.0 final release.

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena